SuLingGG/OpenWrt-Rpi #9: 竞斗云CTCGFW (5.4)版本无线无法发出信号

Benjamin23333, 2020-04-21:
今天刷了给我份竞斗云刷了CTCGFW (5.4)版，在网页管理页面看来无线一些正常，无线的灯也是亮着的，但是终端设备无法接收到无线信号，因为有问题我就没再继续使用，换刷了Lean (4.x)的版本，不清楚还有没其他问题

SuLingGG, 2020-04-21:
> 今天刷了给我份竞斗云刷了CTCGFW (5.4)版，在网页管理页面看来无线一些正常，无线的灯也是亮着的，但是终端设备无法接收到无线信号，因为有问题我就没再继续使用，换刷了Lean (4.x)的版本，不清楚还有没其他问题

这个版本在今天刚刚上线，目前确实存在一些问题，现正尝试修复，待修复后会再次在此 Issues 中回复你~

Benjamin23333, 2020-04-21:
> > 今天刷了给我份竞斗云刷了CTCGFW (5.4)版，在网页管理页面看来无线一些正常，无线的灯也是亮着的，但是终端设备无法接收到无线信号，因为有问题我就没再继续使用，换刷了Lean (4.x)的版本，不清楚还有没其他问题
> 
> 这个版本在今天刚刚上线，目前确实存在一些问题，现正尝试修复，待修复后会再次在此 Issues 中回复你~

好的

xiaogtiang, 2020-04-24:
用的4.19.115版本，中继的信号，不像以前可以设置2.4g和5g信号2合1，要是信号名字一样会没网络。2.4g信号和5g同时设置好了后，在断电或者重启后无线中继不会启动，有信号直接断网

SuLingGG, 2020-04-24:
> 今天刷了给我份竞斗云刷了 CTCGFW (5.4) 版，在网页管理页面看来无线一些正常，无线的灯也是亮着的，但是终端设备无法接收到无线信号，因为有问题我就没再继续使用，换刷了 Lean (4.x) 的版本，不清楚还有没其他问题

理论上修复了这个问题，如果有 5.4 内核需求的话可以测试一下。

SuLingGG, 2020-04-24:
> 用的 4.19.115 版本，中继的信号，不像以前可以设置 2.4g 和 5g 信号 2 合 1，要是信号名字一样会没网络。2.4g 信号和 5g 同时设置好了后，在断电或者重启后无线中继不会启动，有信号直接断网

这个应该是源码的问题...我这边没有做什么大的改动...

Benjamin23333, 2020-04-25:
> > 今天刷了给我份竞斗云刷了 CTCGFW (5.4) 版，在网页管理页面看来无线一些正常，无线的灯也是亮着的，但是终端设备无法接收到无线信号，因为有问题我就没再继续使用，换刷了 Lean (4.x) 的版本，不清楚还有没其他问题
> 
> 理论上修复了这个问题，如果有 5.4 内核需求的话可以测试一下。

我试了，无线真的修复了，暂时没发现其他问题

SuLingGG, 2020-04-26:
> 我试了，无线真的修复了，暂时没发现其他问题
好的，感谢测试~~

Benjamin23333, 2020-04-27:
> > 我试了，无线真的修复了，暂时没发现其他问题
> > 好的，感谢测试~~

我还有个问题一直没说，只不过这回是Lean (4.x)，机型还是竞斗云，因为lean的源码昨天有更新，Project项目并没有同步，我又试了Lean (4.x)。
问题是“未保存的配置”会莫名其妙出现，而且是关于没有进行设置更改的项目；更改某些项目设置并保存后，仍会出现“未保存的项目”，再次保存后仍然出现，只有放弃更改才会消失，且更改会失效，变回默认配置

Benjamin23333, 2020-04-30:
> > 我试了，无线真的修复了，暂时没发现其他问题
> > 好的，感谢测试~~

我后来提的问题解决了么？lean版的问题

SuLingGG, 2020-04-30:
> 我后来提的问题解决了么？lean版的问题
抱歉没看到最后一条...我这儿暂时没测试，也没有收到类似的反馈，我建议你不保留配置刷一下固件试一下。

Benjamin23333, 2020-04-30:
> > 我后来提的问题解决了么？lean版的问题
> 
> 抱歉没看到最后一条...我这儿暂时没测试，也没有收到类似的反馈，我建议你不保留配置刷一下固件试一下。

我从opboot刷的ubi，也不保留配置刷bin试过了，还是这样，难道是玄学问题？

SuLingGG, 2020-04-30:
> 我从opboot刷的ubi，也不保留配置刷bin试过了，还是这样，难道是玄学问题？

我明天测试一下吧...

Benjamin23333, 2020-04-30:
> > 我从opboot刷的ubi，也不保留配置刷bin试过了，还是这样，难道是玄学问题？
> 
> 我明天测试一下吧...

感谢

Benjamin23333, 2020-05-01:
> > 我从opboot刷的ubi，也不保留配置刷bin试过了，还是这样，难道是玄学问题？
> 
> 我明天测试一下吧...

我试了今天lean和project的V1.4，设备是竞斗云，lean版问题是反复自动重启，project版的问题是刷ubi固件上传完成后无法更新，猜测可能是文件体积太大

SuLingGG, 2020-05-01:
> 我试了今天 lean 和 project 的 V1.4，设备是竞斗云，lean 版问题是反复自动重启，project 版的问题是刷 ubi 固件上传完成后无法更新，猜测可能是文件体积太大

我下午测试确实存在这个问题，你如果想找一个暂时稳定的竞斗云固件可以用这个:
https://c-t.work/s/a19eebfac39444

应该是源码的锅，Lean 版源码凌晨编译的竞斗云干脆没有信号，下午的固件反复自动重启，先用这个过渡一下吧~

Benjamin23333, 2020-05-01:
> > 我试了今天 lean 和 project 的 V1.4，设备是竞斗云，lean 版问题是反复自动重启，project 版的问题是刷 ubi 固件上传完成后无法更新，猜测可能是文件体积太大
> 
> 我下午测试确实存在这个问题，你如果想找一个暂时稳定的竞斗云固件可以用这个:
> https://c-t.work/s/a19eebfac39444
> 
> 应该是源码的锅，Lean 版源码凌晨编译的竞斗云干脆没有信号，下午的固件反复自动重启，先用这个过渡一下吧~

好，我试一下

SuLingGG, 2020-05-01:
> 好，我试一下，对了，project 版的问题后续也会解决吗？

应该是源码的锅...我只是个编译爱好者 (编译机器)，只能力所能及地解决一些问题，上游源码发生变动的话我也没办法...
另外从另外一个角度来说，固件是用来满足日常使用需求的，固件运行正常其实是没必要更新的~

SuLingGG, 2020-05-07:
重启问题已经修复了，5.4 和 Lean 的无线中继据测试也应该修复了，如果需要的话可以刷下新固件测试一下。 @Benjamin23333

SuLingGG, 2020-05-11:
此 Issue 将会被关闭，此 Issue 若有后续更新将会再次开启。

closed 2020-05-11 by SuLingGG