SuLingGG/OpenWrt-Rpi #10: 大佬能指点一下我怎么集成docker到固件里么？

guitoushayan, 2020-05-01:
我拉取lean大最新固件，编译时选择树莓派4b后不显示luci-app-dockerman这个选项，但是文件夹里可以搜的到，另外我重新下载luci-app-dockerman源码包(https://github.com/lisaac/luci-app-dockerman)复制到文件夹下之后，有luci-app-dockerman这个选项了
[image]
，但是不能选择上docker-ce，一选择docker-ce，luci-app-dockerman自动消失
[image]另外我编译完的固件docker项目下只有一个概况
[image]

SuLingGG, 2020-05-02:
首先删除存放在 packages 文件夹下 lisaac/luci-app-dockerman 的项目文件夹，删除完毕之后:

```
cd package
git clone --depth=1 https://github.com/KFERMercer/luci-app-dockerman
mkdir luci-lib-docker
curl -s -o ./luci-lib-docker/Makefile https://raw.githubusercontent.com/lisaac/luci-lib-docker/master/Makefile
rm -rf ../lean/luci-app-dockerman
```

之后回到 OpenWrt 源码根目录， `make menuconfig` 即可。

如果问题还未解决，在 OpenWrt 源码根目录执行: `rm -rf ./tmp && rm .config` (此操作会删除你 make menuconfig 生成的配置)，之后重新 `make menuconfig` 即可。

guitoushayan, 2020-05-03:
感谢大佬指点，已经成功集成进固件了~

closed 2020-05-03 by guitoushayan