SuLingGG/OpenWrt-Rpi #17: 无法使用pip/pip3 同时使用官方编译的固件无法安装rpi.gpio模块

liangfyx, 2020-07-22:
python-pip3
```
Traceback (most recent call last):
  File "/usr/bin/pip3", line 7, in <module>
    from pip._internal import main
  File "/__init__.py", line 40, in <module>
  File "/autocompletion.py", line 8, in <module>
  File "/main_parser.py", line 12, in <module>
  File "/__init__.py", line 6, in <module>
  File "/completion.py", line 6, in <module>
  File "/base_command.py", line 18, in <module>
  File "/download.py", line 38, in <module>
  File "/glibc.py", line 3, in <module>
  File "/__init__.py", line 7, in <module>
ImportError: Error loading shared library libffi.so.6: No such file or directory (needed by /usr/lib/python3.6/lib-dynload/_ctypes.cpython-36.so)
```
python-pip
```
ERROR:root:code for hash md5 was not found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/hashlib.py", line 147, in <module>
  File "/usr/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
ValueError: unsupported hash type md5
ERROR:root:code for hash sha1 was not found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/hashlib.py", line 147, in <module>
  File "/usr/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
ValueError: unsupported hash type sha1
ERROR:root:code for hash sha224 was not found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/hashlib.py", line 147, in <module>
  File "/usr/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
ValueError: unsupported hash type sha224
ERROR:root:code for hash sha256 was not found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/hashlib.py", line 147, in <module>
  File "/usr/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
ValueError: unsupported hash type sha256
ERROR:root:code for hash sha384 was not found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/hashlib.py", line 147, in <module>
  File "/usr/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
ValueError: unsupported hash type sha384
ERROR:root:code for hash sha512 was not found.
Traceback (most recent call last):
  File "/usr/lib/python2.7/hashlib.py", line 147, in <module>
  File "/usr/lib/python2.7/hashlib.py", line 97, in __get_builtin_constructor
ValueError: unsupported hash type sha512
Traceback (most recent call last):
  File "/usr/bin/pip", line 6, in <module>
    from pip._internal import main
  File "/usr/lib/python2.7/site-packages/pip/_internal/__init__.py", line 19, in <module>
  File "/usr/lib/python2.7/site-packages/pip/_vendor/urllib3/__init__.py", line 7, in <module>
  File "/usr/lib/python2.7/site-packages/pip/_vendor/urllib3/connectionpool.py", line 30, in <module>
  File "/usr/lib/python2.7/site-packages/pip/_vendor/urllib3/connection.py", line 39, in <module>
  File "/usr/lib/python2.7/site-packages/pip/_vendor/urllib3/util/__init__.py", line 6, in <module>
  File "/usr/lib/python2.7/site-packages/pip/_vendor/urllib3/util/ssl_.py", line 8, in <module>
ImportError: cannot import name md5
```

SuLingGG, 2020-07-22:
我不是很建议在 OpenWrt 上使用 Python 和 GPIO，因为 OpenWrt 是面向网络设备和嵌入式设备的一个 Linux 发行版，不是面向生产环境的比较标准的 Linux 发行版，一些情况下，无法像 Ubuntu 或者 CentOS 那样很完美地实现一些特性。

另外，如果你是将 OpenWrt 用作旁路由使用（几乎用不到树莓派的无线功能），那么我推荐你以 Raspbian 或者 Ubuntu 这类比较成熟的 Linux 发行版做宿主系统，在宿主系统中安装 Docker 并且尝试一下我的 Docker 版 OpenWrt：

<https://mlapp.cn/376.html>

<https://github.com/SuLingGG/OpenWrt-Rpi-Docker>

同时，在宿主系统中安装和运行你需要的 Python 模块和程序。宿主系统对 Python 的支持应该大大好过 OpenWrt。

另外，在 OpenWrt 中我已预编译了 gpioctl-sysfs 和 gpiod-tools 两个组件，不知能否满足你的需求~

SuLingGG, 2020-07-23:
如果您未回复，该 Issue 将在 24 小时内关闭~

closed 2020-07-26 by SuLingGG