SuLingGG/OpenWrt-Rpi #21: 关于X86_64镜像 iptables 问题

penghaiheng, 2020-08-17:
我看到docker源那里已经更新了, X86_64的镜像, 所以就尝试装在了docker里, ssrplus 因为看不到具体日志, 所以就用 passwall 添加节点了, 但是之后在日志里一直看到有
'''
开始加载防火墙规则...
系统不兼容，终止执行！
重启 dnsmasq 服务[0]
运行完成！
'''
这样的信息, 我自己尝试配置iptables , 是可以正常执行的. 同时后台也看不到dnsmasq的进程了. 手动启动dnsmasq是可以的. 烦请解惑了. 感谢. 因为实在不确定防火墙规则的具体内容, 所以只能搁置在那里.

penghaiheng, 2020-08-18:
之后又看了一下 ipset , 发现 ipset list 的时候, 会直接报错. "ipset v7.4: Cannot open session to kernel."  可能是因为内核编译使用的 kmod-ipt-ipset 比内核版本要高吧 ?

SuLingGG, 2020-08-18:
我觉得这个和宿主机的内核配置有关系，因为 Docker 容器和宿主机共享内核，所以在 Docker 中运行的 OpenWrt 也是使用宿主机的内核，可能宿主机的内核没法完全兼容 OpenWrt 的内核需求，所以会报错。

我这边测试是没有问题的，宿主机内核为 5.7.14-1-MANJARO。

下面是我在 x86_64 设备上在 Docker 中运行 OpenWrt 的运行截图(使用 ShadowsocksR Plus+ )，设备用 Docker 中的 OpenWrt 也可以正常使用科学上网。

[image]

closed 2020-08-21 by SuLingGG