SuLingGG/OpenWrt-Rpi #23: 树莓派4b使用网件A6210做AP断连

nakamakaede, 2020-09-10:
你好，我的配置是：
树莓派4b，Lean 大源码，无线网卡：网件A6210
在开启5gWiFi做热点时，进行测速或看视频时，每次都会断开连接，重新连接WiFi时显示“不能连接到这个网络”，重启树莓派后能重新连接。但看视频时又会掉线。请问应该怎么解决？谢谢。
附：两次断开连接后系统日志
第一次：
`Thu Sep 10 21:03:14 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 20:0d:b0:46:0c:08
Thu Sep 10 21:03:14 2020 daemon.info hostapd: wlan0: STA 20:0d:b0:46:0c:08 IEEE 802.11: disassociated
Thu Sep 10 21:03:14 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 20:0d:b0:46:0c:08 1
Thu Sep 10 21:03:15 2020 daemon.info hostapd: wlan0: STA 20:0d:b0:46:0c:08 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)`
第二次：
`Thu Sep 10 22:39:55 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Thu Sep 10 22:39:55 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Thu Sep 10 22:39:58 2020 kern.err kernel: [  217.677072] ieee80211 phy1: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55447, -52
Thu Sep 10 22:40:04 2020 kern.err kernel: [  223.696675] ieee80211 phy1: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55447, -52
Thu Sep 10 22:40:06 2020 kern.err kernel: [  225.538318] mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -110
Thu Sep 10 22:40:10 2020 kern.err kernel: [  229.715511] ieee80211 phy1: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55447, -52
Thu Sep 10 22:40:15 2020 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Thu Sep 10 22:40:15 2020 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Thu Sep 10 22:40:15 2020 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Thu Sep 10 22:40:16 2020 kern.err kernel: [  235.738023] ieee80211 phy1: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55447, -52
Thu Sep 10 22:40:22 2020 kern.err kernel: [  241.759303] ieee80211 phy1: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55447, -52
Thu Sep 10 22:40:28 2020 kern.err kernel: [  247.778299] ieee80211 phy1: brcmf_cfg80211_start_ap: Set Channel failed: chspec=55447, -52
Thu Sep 10 22:40:30 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 20:0d:b0:46:0c:08
Thu Sep 10 22:40:30 2020 daemon.info hostapd: wlan0: STA 20:0d:b0:46:0c:08 IEEE 802.11: disassociated
Thu Sep 10 22:40:30 2020 daemon.notice hostapd: wlan0: STA-OPMODE-N_SS-CHANGED 20:0d:b0:46:0c:08 1
Thu Sep 10 22:40:31 2020 daemon.info hostapd: wlan0: STA 20:0d:b0:46:0c:08 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Sep 10 22:40:32 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED d8:c4:97:e6:45:5a`

nakamakaede, 2020-09-10:
对了，2.4gWiFi连接正常

SuLingGG, 2020-09-10:
另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。

nakamakaede, 2020-09-10:
好的，谢谢您，我去试一下

nakamakaede, 2020-09-10:
> 另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。

我试了一下，还是不行啊，刷了最新的固件还是有这个问题

SuLingGG, 2020-09-10:
> > 另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。
> 
> 我试了一下，还是不行啊，刷了最新的固件还是有这个问题

你如果按我上面的做了一遍还是有这个问题的话，检查一下树莓派供电是否正常 (红灯是否闪烁)，并且把网卡插到靠近网口上部的那个 USB 再试一下。

nakamakaede, 2020-09-11:
> > > 另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。
> > 
> > 
> > 我试了一下，还是不行啊，刷了最新的固件还是有这个问题
> 
> 你如果按我上面的做了一遍还是有这个问题的话，检查一下树莓派供电是否正常 (红灯是否闪烁)，并且把网卡插到靠近网口上部的那个 USB 再试一下。

我检查了一下，红灯常亮，网卡是靠近网口的上面的那个usb3.0。
我突然发现，当我今天只用平板电脑连接5gWiFi时，信号挺稳定的。但是我昨天同时使用了笔记本和平板两个设备连接的WiFi，出现了断线问题。所以我推测有可能是同时连接了5g WiFi互相干扰的原因。
我今晚和明天再测试一下，如果是这个原因的话我就跟您确认一下，应该就可以close issue了。
非常感谢您的帮助。谢谢！！！

nakamakaede, 2020-09-11:
> > > 另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。
> > 
> > 
> > 我试了一下，还是不行啊，刷了最新的固件还是有这个问题
> 
> 你如果按我上面的做了一遍还是有这个问题的话，检查一下树莓派供电是否正常 (红灯是否闪烁)，并且把网卡插到靠近网口上部的那个 USB 再试一下。

哦。。。我都要放弃了，我刚回复完，WiFi就断了。。。

nakamakaede, 2020-09-11:
> > > 另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。
> > 
> > 
> > 我试了一下，还是不行啊，刷了最新的固件还是有这个问题
> 
> 你如果按我上面的做了一遍还是有这个问题的话，检查一下树莓派供电是否正常 (红灯是否闪烁)，并且把网卡插到靠近网口上部的那个 USB 再试一下。
我看见了发生错误时的日志 ：
`Fri Sep 11 20:58:13 2020 kern.err kernel: [  442.950157] mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -110
Fri Sep 11 20:58:33 2020 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED d8:c4:97:e6:45:5a`

nakamakaede, 2020-09-11:
> 另外如果你 A6210 的 SSID 还是 OpenWrt 的话，我建议你改个名称，因为树莓派的板载无线的 SSID 也是 OpenWrt，且为 5G 频段，如果是你的设备用着用着连接到了树莓派的板载无线没准也会导致这个问题。
>
>
> 我试了一下，还是不行啊，刷了最新的固件还是有这个问题
>
>
> 你如果按我上面的做了一遍还是有这个问题的话，检查一下树莓派供电是否正常 [...]

SuLingGG, 2020-09-11:
也许你可以试试这个：

https://github.com/openwrt/mt76/issues/405#issuecomment-665737497

这个 issue 里面说这条错误：
```
mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -110

可能只会在网卡连接一台以上设备的时候才会发生。
待会儿要上夜班，我明天下午测试一下。

nakamakaede, 2020-09-11:
> 也许你可以试试这个：
> 
> [openwrt/mt76#405 (comment)](https://github.com/openwrt/mt76/issues/405#issuecomment-665737497)
> 
> 这个 issue 里面说这条错误：
> 
> ```
> mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -110
> ```
> 
> 可能只会在网卡连接一台以上设备的时候才会发生。
> 
> 解决方法：
> 拔下 USB 网卡，运行此命令，插入 USB 网卡:
> 
> ```
> echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg
> ```
> 
> 待会儿要上夜班，我明天下午测试一下。

好的，我试试。您先忙，非常感谢您的帮助

nakamakaede, 2020-09-12:
> 也许你可以试试这个：
>
> [openwrt/mt76#405 (comment)](https://github.com/openwrt/mt76/issues/405#issuecomment-665737497)
>
> 这个 issue 里面说这条错误：
>
> ```
> mt76x2u 2-1:1.0: error: mt76x02u_mcu_wait_resp failed with -110 [...]

我试了一下
使用了
`echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg`
这个命令之后确实好使了
用两个设备同时播放了一个小时的4k视频，没有出现问题。
就是每次重启后都需要重新运行一下这个命令。
谢谢您的帮助！

SuLingGG, 2020-09-12:
> 我试了一下
> 使用了
> `echo 1 > /sys/module/mt76_usb/parameters/disable_usb_sg`
> 这个命令之后确实好使了
> 用两个设备同时播放了一个小时的4k视频，没有出现问题。
> 就是每次重启后都需要重新运行一下这个命令。
> 谢谢您的帮助！

可以试试:

新建文件: 
/etc/modules.d/mymoduleparam.conf

并在其中添加以下内容:
mt76-usb disable_usb_sg=1

然后保存并重启，重启之后:
```
cat /sys/module/mt76_usb/parameters/disable_usb_sg
```
如果返回结果是1那么就应用成功了，之后每次重启不必每次都执行此命令。

我会把这个配置加进明天的固件里，这样以后的固件就能实现开机即用且无需另外手动配置了。

nakamakaede, 2020-09-12:
> 可以试试:
>
> 新建文件:
> /etc/modules.d/mymoduleparam.conf
>
> 并在其中添加以下内容:
> mt76-usb disable_usb_sg=1
>
> 然后保存并重启，重启之后:
>
> ```
> cat /sys/module/mt76_usb/parameters/disable_usb_sg
> ```
>
> 如果返回结果是1那么就应用成功了，之后每次重启不必每次都执行此命令。 [...]

太好了，谢谢您

closed 2020-09-12 by nakamakaede