SuLingGG/OpenWrt-Rpi #25: 发现树莓派3B+插着网线启动或者重启获取不到ip

qrcode-star, 2020-09-26:
发现树莓派3B+插着网线启动或者重启获取不到ip（或者根本没有启动，无法验证）。只有先插电源等待15到20秒以后再插网线，才可以正常获取IP，正常访问。

qrcode-star, 2020-09-26:
lan口ip无论设置成静态还是dhcp都不行

SuLingGG, 2020-09-26:
你反映的这个情况之前没有人反馈过，我自己也没有遇到这个问题...
建议删除所有分区，新建一个 Fat32 分区，重刷一下固件试试。
并且如果你当前使用的是 squashfs 固件，那么这次刷 ext4 固件试试，反之亦然。

irol765, 2020-09-27:
https://www.ichenfu.com/2019/09/14/raspberrypi-boot-without-HDMI-connected/

解药

irol765, 2020-09-27:
我也遇到了，反映下。

SuLingGG, 2020-09-28:
> 我也遇到了，反映下。

重启之后可以搜索到树莓派发射出来的热点吗？

qrcode-star, 2020-10-06:
没有时间验证此问题的具体原因。我已在树莓派刷入debian镜像，然后使用docker运行openwrt，比较稳定，此问题关闭吧

SuLingGG, 2020-10-06:
> 没有时间验证此问题的具体原因。我已在树莓派刷入 debian 镜像，然后使用 docker 运行 openwrt，比较稳定，此问题关闭吧

可以试一下这篇文章的解决方法：

yt880924, 2020-10-09:
> > 没有时间验证此问题的具体原因。我已在树莓派刷入 debian 镜像，然后使用 docker 运行 openwrt，比较稳定，此问题关闭吧
> 
> 可以试一下这篇文章的解决方法：
> https://www.ichenfu.com/2019/09/14/raspberrypi-boot-without-HDMI-connected/

我也遇到同样的问题,只要重启就无法启动,每次都需要拔掉网线,等一会儿再插上才可以.
已经按照上面的教程设置,但是仍然不行.

SuLingGG, 2020-10-09:
> > > 没有时间验证此问题的具体原因。我已在树莓派刷入 debian 镜像，然后使用 docker 运行 openwrt，比较稳定，此问题关闭吧
> > 
> > 
> > 可以试一下这篇文章的解决方法：
> > https://www.ichenfu.com/2019/09/14/raspberrypi-boot-without-HDMI-connected/
> 
> 我也遇到同样的问题，只要重启就无法启动，每次都需要拔掉网线，等一会儿再插上才可以.
> 已经按照上面的教程设置，但是仍然不行.

抱歉我手头没有 3B+ 无法测试...
如果方便的话，可以试试重启后连接显示器和键盘，在启动过程中会在屏幕上打印出内核日志。
启动完成 (内核日志不再滚动时) 后按 Enter 键即可进入树莓派的终端，执行以下命令：

```
dmseg > /root/dmseg.txt
```

这样即可将此次启动过程中的内核日志输出至 /root/dmseg.txt 文件中，之后拔掉网线再插入，正常启动 OpenWrt 后下载此文件到本地，然后上传此文件到这里，我看看能不能找出问题来...

whispery-project, 2020-10-11:
> 抱歉我手头没有 3B+ 无法测试...
> 如果方便的话，可以试试重启后连接显示器和键盘，在启动过程中会在屏幕上打印出内核日志。
> 启动完成 (内核日志不再滚动时) 后按 Enter 键即可进入树莓派的终端，执行以下命令：
>
> ```
> dmseg > /root/dmseg.txt
> ```
>
> 这样即可将此次启动过程中的内核日志输出至 /root/dmseg.txt 文件中，之后拔掉网线再插入，正常启动 [...]

我目前解决了问题,截至目前在后台手动重启过两次,没有出现问题,**定时重启还未测试**.
下面详细写一下我遇到的问题和解决问题的的过程.

我的主路由是AC86U[koolshare梅改固件],树莓派B3+刷的这边OpenWRT固件.
之前主路由版本 **[384.18_0往前](https://koolshare.cn/thread-127878-1-1.html#3841802020%E5%B9%B407%E6%9C%8802%E6%97%A5)**
期间更新过很多次主路由固件,也在用这边的OpenWRT固件跟路由基本同时更新.
始终出现每次重启,**手动和定时**,都需要拔掉网线,等一会儿再插上才可以恢复正常.
除了上述问题没有出现其它问题,我就没折腾.

当我最近将主路由固件升级至 **[384.19_0](https://koolshare.cn/thread-127878-1-1.html#3841902020%E5%B9%B408%E6%9C%8821%E6%97%A5)**
因为主路由软件中心出现问题,将主路由进行了双清. **恢复出厂设置+格式化jffs分区**
之后我又重新刷了这边的OpenWRT,做旁路由,仍然会出现上面所提到的**重启问题**,而且还增加了新的问题.
新增的问题就是 _有线连接可以访问网络_,**无线设备无法访问网络**.

之后看了这边的[旁路由设置Wiki](https://github.com/SuLingGG/OpenWrt-Rpi/wiki/%E6%97%81%E8%B7%AF%E7%94%B1%E8%AE%BE%E7%BD%AE)
里面的条目都有逐一测试.
只有在OpenWRT后台添加防火墙规则这一条有效.
**_iptables -t nat -I POSTROUTING -j MASQUERADE_**
但是添加防火墙规则后,**Wifi内网速度大幅受限**,**访问国内网站速度大幅下降**,_Wifi外网不受影响_,_有线连接均不受影响_.
**而且重启问题仍然存在.**

之后在网上找了很多相关的资料.
最后参考了这篇[**帖子**](https://koolshare.cn/thread-166096-1-1.html).
**主要参考了**[**15楼**](https://koolshare.cn/forum.php?mod=redirect&goto=findpost&ptid=166096&pid=2181325).
[**16楼**](https://koolshare.cn/forum.php?mod=redirect&goto=findpost&ptid=166096&pid=2182427),提到的问题也有一部分能对上,**但我没有参考16楼的设置**.

之后我将**主路由AC86U的AiProtection开启并重启路由器**._**[主路由固件版本384.19_0更新前,AiProtection均未开启.]**_
树莓派B3+重新刷入这边的OpenWRT固件,**防火墙规则默认,其它设置按Wiki照搬**,**并重启OpenWRT**,**目前恢复正常**.
等使用一段时间之后再来反馈,定时重启的问题,就这样.
最后感谢SuLing提供的固件.

bookyue, 2020-10-14:
@whispery-project Maybe you should check out this.
[RPi hangs and crashes with lowered idle/min clock on Linux 5.4.51+](https://github.com/MichaIng/DietPi/issues/3690)
I haven't try this OpenWRT firmware(master branch) yet. If the kernel version of latest OpenWRT firmware is 5.4.51+.
That make sense.

sjzdsong, 2020-12-05:
> 发现树莓派3B +插着网线启动或重启获取不到ip（或者根本没有启动，无法验证）。只有先插电源等待15到20秒以后再插网线，才可以正常获取IP，正常访问。

我3b+也是重启后无法获得地址。

LycmSky, 2021-01-14:
我也有同样的情况

LycmSky, 2021-01-14:
> 我也有同样的情况

换squashfs版固件以后好像可以了,手动重启两次以及直接断电一次都成功连上了

aweigz, 2021-01-18:
也是有同样的问题，需要断电重启N次，另外，进入系统后修改过的配置，经常无法保存

1648553305, 2021-01-23:
不知道是不是同样的问题，我遇到的是：刷完系统之后，配置好wan口为dhcp客户端之后，保存完应用可以正常上网了，重启一下就会发现wan口是无法获取ip地址的，只能重刷系统，我尝试过重启网卡，拔插网线，都没有解决，也有更换内存卡刷个4g的覆盖镜像，似乎没有解决，而且lean版和officia版的固件都存在这种问题，奇怪的是如果是外接的usb有线网卡没有这问题。

1648553305, 2021-01-27:
> 不知道是不是同样的问题，我遇到的是：刷完系统之后，配置好wan口为dhcp客户端之后，保存完应用可以正常上网了，重启一下就会发现wan口是无法获取ip地址的，只能重刷系统，我尝试过重启网卡，拔插网线，都没有解决，也有更换内存卡刷个4g的覆盖镜像，似乎没有解决，而且lean版和officia版的固件都存在这种问题，奇怪的是如果是外接的usb有线网卡没有这问题。

我前两天用得好好的，今天插上电发现又获取不了IP了，我检查了一下，发现网口是不运行的，运行时间为0，我就在想会不会是电源的问题，导致网口供电不足，再加上树莓派的电源指示灯我看着不太对劲，于是我换了一条usb线，于是就解决了。有相同问题的小伙伴可以试一下，先看看电源指示灯（红色的那个）是不是常亮，如果不是，很有可能电源有问题导致自带的网卡不工作，WAN口会显示运行时间为0，进而表现为无法获取IP

DBWyouzi, 2021-01-28:
会不会是因为设置了lan里面不获取dhcp（忽略此接口/不在此接口提供 DHCP 服务）的缘故啊

DBWyouzi, 2021-01-31:
> 会不会是因为设置了lan里面不获取dhcp（忽略此接口/不在此接口提供 DHCP 服务）的缘故啊

我是用的重建一个WAN＋DHCP来获取我主路由器的ip，然后再转发到lan接口，用这种方法我重启都不会出现连接但获取不了ip的情况。但我一旦修改lan接口上的配置（及勾上忽略此接口/不在此接口提供DHCP服务）的时候重启3B+的WIFI就一直处于连接但连不上的问题（有可能就是分配不到IP导致的）

aka714, 2021-02-06:
> 先插电源等待15到20秒以后再插网线，才可以正常获取IP，正常访问

这个问题我遇到过：【插着网线启动各种不正常（包括不能获取IP），而不插网线通电启动，再等个几十秒后再插网线就啥毛病没有】

初步尝试：
1）插显示器，并测树莓派4B的输入功率、电压和电流，发现是树莓派4B供电不正常（10次有个几次会自动重启或死机）；
2）检查适配器，功率足够没有问题；
3）发现问题：是我用了根铜线很细的垃圾Type-C线，这根线是个小夜灯的充电线，只带得动功率2、3瓦的设备，电流一大压降就大，树莓派供电不够就重启或不正常了；
4）不插网线启动，树莓派不用处理太多东东，启动的时候负载小，因此成功启动的概率高。

解决方法：
1）换了一根优质Type-C充电线；
2）搞定。

后续测试，再尝试:
a）换回前一根垃圾线，插着网线启动10次，结果9次不正常；
b）换回前一根垃圾线，先通电启动，等个几十秒，然后再插网线，10次，结果2次不正常；
c）使用优质线，插着网线启动和先启动再插网线，两种方式各尝试10次，20次结果均正常。

aka714, 2021-02-06:
> 我前两天用得好好的，今天插上电发现又获取不了IP了，我检查了一下，发现网口是不运行的，运行时间为0，我就在想会不会是电源的问题，导致网口供电不足，再加上树莓派的电源指示灯我看着不太对劲，于是我换了一条usb线，于是就解决了。有相同问题的小伙伴可以试一下，先看看电源指示灯（红色的那个）是不是常亮，如果不是，很有可能电源有问题导致自带的网卡不工作，WAN口会显示运行时间为0，进而表现为无法获取IP

正解！我遇到的一样，换根优质充电线，搞定！

nemo2018, 2021-02-10:
发现了一模一样的问题。同样的3B+，静态IP，带着网线冷启动或者控制台执行重启，就连不上了。不带网线启动，等一分钟再连上网线就没问题。
换上了质量很好的线和很好的高功率5V电源，没有改善。
在squashfs和ext4两种系统都试过，现象一样。