SuLingGG/OpenWrt-Rpi #28: Universal plug-in leaking DNS / 通用插件泄漏DNS

taricano, 2020-10-09:
I ran a dns test while using a v2ray on the universal plug-in shadowsocksR + and there was a leak. Is there anyone else having this issue? 

我在通用插件shadowsocksR+上使用v2ray时运行了一个dns测试，结果出现了漏洞。还有其他人有这个问题吗？

SuLingGG, 2020-10-09:
我不是网络安全专家...不是很清楚你要描述的问题。
(包括上文提到的 Dns 测试 / 漏洞)

此项目 Issues 仅讨论固件自身存在的问题和改进建议。

如果你要反馈源码中存在的问题，请移步：

ShadowsocksR Plus + 所在项目：
<https://github.com/fw876/helloworld>
<https://github.com/coolsnowwolf/lede>

V2ray 所在项目：
<https://github.com/v2fly/v2ray-core>

taricano, 2020-10-09:
理解并感谢提供链接。你的固件很棒，我很喜欢用它。

bailaowai, 2020-10-14:
I figured it out. On the wan interface into advance options, an alternative DNS must be used. This eliminates the leaks completely.

SuLingGG, 2020-10-24:
Well, Find the problem is OK...

closed 2020-10-24 by SuLingGG

bailaowai, 2020-10-24:
I did it.

Go to interfaces,
Network,
On wan (eth1),
Advanced settings,
Unselect Use DNS servers advertised by peer,
Enter 208.67.222.222, click on the plus sign ➕ to add it,
Do the same thing as above but enter 208.67.220.220,
You can use any other DNS public servers you want but these two work well
in China, you can also use 119.29.29.29 & 182.254.116.116 or
114.114.114.114 & 114.114.115.115. (Although these DNSs are Chinese, there
is no issue about leaks by using it into this mode).
Click Save & Apply.

Under services, make sure you are using IP Route Mode or Global Mode
in the ShadowSocks
R Plus + plug-in. Also, the anti pollution DNS Server must be any of the
non Chinese ones, otherwise you will get your DNS exposed or polluted.

Save and apply all the settings, restart your router and flush your DNS on
your computer by executing ipconfig /flushdns on a command line.

Run a DNS test on www.dnsleaktest.com

There should not be any Chinese routing there.

This is very important because if there is a leak, it is likely that
whatever serves you are using as proxy will be blocked eventually.