SuLingGG/OpenWrt-Rpi #29: nginx 增加http_ssl_module 以支持HTTPS

illnino, 2020-10-15:
## 问题

现在固件中内置的nginx不包含`http_ssl_module`, 无法搭建https 网站。

```
$ nginx -V
nginx version: nginx/1.19.0 (x86_64-pc-linux-gnu)
configure arguments: --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix= --program-suffix= --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls --crossbuild=Linux::aarch64 --prefix=/usr --conf-path=/etc/nginx/nginx.conf --add-module=/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-no-ssl/nginx-1.19.0/nginx-naxsi/naxsi_src --with-ipv6 --add-module=/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-no-ssl/nginx-1.19.0/nginx-ubus-module --add-module=/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-no-ssl/nginx-1.19.0/nginx-headers-more --error-log-path=stderr --pid-path=/var/run/nginx.pid --lock-path=/var/lock/nginx.lock --http-log-path=/var/log/nginx/access.log --http-client-body-temp-path=/var/lib/nginx/body --http-proxy-temp-path=/var/lib/nginx/proxy --http-fastcgi-temp-path=/var/lib/nginx/fastcgi --with-cc=aarch64-openwrt-linux-musl-gcc --with-cc-opt='-I/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -I/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/include -I/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include/fortify -I/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/include -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/build_dir/target-aarch64_cortex-a53_musl/nginx-no-ssl/nginx-1.19.0=nginx-1.19.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fvisibility=hidden -ffunction-sections -fdata-sections -DNGX_LUA_NO_BY_LUA_BLOCK' --with-ld-opt='-L/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/target-aarch64_cortex-a53_musl/lib -L/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/home/runner/work/OpenWrt-Rpi/OpenWrt-Rpi/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -znow -zrelro -Wl,--gc-sections' --without-http_upstream_zone_module
```

Ref: http://nginx.org/en/docs/http/ngx_http_ssl_module.html

### Nginx Conf

```
server {
    listen 443 ssl http2 ;
    server_name mb3admin.com;
    ssl_certificate '/root/burp/ca-uk.der';
    ssl_certificate_key '/root/burp/server-uk.key';
    ssl_protocols TLSv1.1 TLSv1.2 TLSv1.3;
    ssl_prefer_server_ciphers   on;
    ssl_session_cache 'shared:SSL:32k';
    ssl_session_timeout '64m';
    add_header Access-Control-Allow-Origin *;
    add_header Access-Control-Allow-Headers *;
    add_header Access-Control-Allow-Method *;
    add_header Access-Control-Allow-Credentials true;
    ...
}

```

### Error Message
```
$ nginx -t
2020/10/15 16:20:49 [emerg] 29684#0: the "ssl" parameter requires ngx_http_ssl_module in /etc/nginx/conf.d/_emby.conf:2
nginx: configuration file /etc/nginx/nginx.conf test failed
```

## 固件

OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)

谢谢！

SuLingGG, 2020-10-17:
如果你想在 OpenWrt 中使用 Nginx，建议通过 Docker 拉取基于 arm/arm64 的 Nginx Docker 镜像。OpenWrt 自带 Nginx 仅为某些软件包的依赖，仅保证满足内置软件包运行需求，对于额外的需求，建议用 Docker 实现。

illnino, 2020-10-17:
Luci 不是运行在Nginx上吗？
如果固件的Nginx在打包时用了http_ssl_module，就可以直接在`/etc/nginx/conf.d` 添加一个conf，重启nginx即可了。

这样是不是更方便？

谢谢

SuLingGG, 2020-10-17:
首先，OpenWrt 的 Luci 是通过 uhttpd 作为 Web 服务器的。
另外，固件内自带的 Nginx 为 luci-app-kodexplorer (可道云) 的依赖。在未启用可道云的服务之前，Nginx 并不会运行。
其次，OpenWrt 的编译方式与单独编译 Nginx 不同，OpenWrt 虽然也有让 Nginx 支持 SSL 的选项，但是编译和测试可能会花费我大量的时间。而让 Nginx 在 Docker 中运行已经为业界成熟的方案，方式更灵活且兼容性更高，所以我目前还是建议你使用 Docker 容器的方式来运行 Nginx。

illnino, 2020-10-17:
明白了，谢谢大佬。

closed 2020-10-17 by illnino