SuLingGG/OpenWrt-Rpi #33: 能否编一个树莓派zero w的版本

CrazyMed, 2020-11-04:
您好 我是一个手里只有 raspberry pi zero w 的蛋疼玩家。
不知道是否能编一个pi0w版本 加入usb to ssh 功能。
感谢。网上有关pi0w的 openwrt 版本太少了

SuLingGG, 2020-11-04:
Zero 以及 Zero W 用树莓派 1 的版本就可以，Zero W 群内已经有多位玩家证实可以正常使用了，Zero 目前还不太清楚。
另外刷入我的固件之后 Zero W 会自动发射出一个名为 OpenWrt 的热点，连接此热点通过 192.168.1.1 即可访问树莓派 OpenWrt 的 LuCI 或 SSH。

CrazyMed, 2020-11-04:
> Zero 以及 Zero W 用树莓派 1 的版本就可以，对于 Zero W，群内已经有多位玩家证实可以正常使用本项目固件了，Zero 目前还不太清楚。
> 另外刷入我的固件之后 Zero W 会自动发射出一个名为 OpenWrt 的热点，连接此热点通过 192.168.1.1 即可访问树莓派 OpenWrt 的 LuCI 或 SSH。

感谢回复。我去试试。辛苦了

CrazyMed, 2020-11-05:
> Zero 以及 Zero W 用树莓派 1 的版本就可以，对于 Zero W，群内已经有多位玩家证实可以正常使用本项目固件了，Zero 目前还不太清楚。
> 另外刷入我的固件之后 Zero W 会自动发射出一个名为 OpenWrt 的热点，连接此热点通过 192.168.1.1 即可访问树莓派 OpenWrt 的 LuCI 或 SSH。

你好试过成功输入了 树莓派的1的版本。

有个需求就是需要通过 zero  w  的usb 管理设备，需要 添加  以下两个包，我查过这两个包 在最新的Lean 大编译版本里 没有，不知道能不能在下个版本中加入？
opkg install kmod-usb-gadget-eth 
opkg install kmod-usb-dwc2 

以下手动编辑即可
echo 'dtoverlay=dwc2' >> /boot/config.txt

closed 2020-11-11 by SuLingGG