SuLingGG/OpenWrt-Rpi #34: 树莓派4B作为AP-client模式时区域码（reg/ccode）异常的问题

cndongle, 2020-11-05:
这个问题好像官方3/3B就有，说法是先用Raspbian设置好，再切回openwrt，但是在4B上好像无效，我手头的v1.1和v1.2硬件版本都有这个问题，不知道您有没有高见。

具体来说，用AP-client模式连接其他wifi，树莓派作为网关（wwan），有线口做内网lan。
这个时候并不能scan到一些如149,153,157等等的5GHz ssid，低段2.4GHz的ssid是可以的，但是36 channel也是可以搜到的。
这个状态感觉和区域码直接相关，iw reg get默认返回00，set CN后scan并不改善，且重启后丢失。

相关讨论：
[https://forum.openwrt.org/t/ac-mode-on-raspberry-pi-4-wifi/65074/37](https://forum.openwrt.org/t/ac-mode-on-raspberry-pi-4-wifi/65074/37)

SuLingGG, 2020-11-06:
听起来这个问题和底层固件有关系，我这边暂时无法解决。
如果你确实有便携需求 (出差/经常住酒店/中继教室 WIFI) 的话，我还是建议你买一块 USB 无线网卡做中继来使用。
我刚刚测试使用网件 A6210 (MT7602U) 的无线网卡可以正常扫描到 149 信道的 SSID。
如果你手头有闲置的路由器，还是建议你用路由器中继 WIFI，然后将树莓派网口与路由器 Lan 口相连，让树莓派做旁路由的方式来使用，这样能获得最好的无线性能和上网体验，同时可以享受到 OpenWrt 带来的绝大多数功能。
不过你既然提出了这个问题，肯定是所处场景不方便或者不想使用路由器的场景。所以我建议你买一块 USB 无线网卡来解决你遇到的问题。
另外网件 A6210 闲鱼 100 以下就能买到，如果可以接受插一个 USB 小尾巴的话可以尝试一下。

cndongle, 2020-11-06:
明白了，感谢。 我的应用场景确实相当奇葩，奇葩到板载这个1x1的网卡都还有余量T.T。
现在是刷了ubuntu，然后lxd起openwrt的容器当软路由，越过了这个问题。单位有comfast CF-917AC（RTL8814AU/8813）,不过没在openwrt下尝试是否可行。

closed 2020-11-06 by cndongle