SuLingGG/OpenWrt-Rpi #39: 试了lede 5.4 x86_64新版

zyystudio, 2020-11-12:
zerotier无法使用，只好卸载，opkg安装新版就可以

SuLingGG, 2020-11-12:
“安装新版” 是指 opkg install zerotier 安装这个软件包吗？

zyystudio, 2020-11-12:
是的 1.4.6-5.但是我感觉是luci问题。还有试了frp内网穿透 也有问题，提示认证密码不匹配，但是保证密码是没错的。因为之前用树莓派3b和4感觉挺好的就把家里主路由的openwrt换成你的了。是不是lede的问题啊？基本主要的都没问题，只是这些个增值项感觉上有些没测试好也没办法测试全面

SuLingGG, 2020-11-12:
我明天用 snapshots 的 zerotier 试试看。另外 frp 应该没问题，如果提示密码不匹配可能是服务端那边的版本太低了。

zyystudio, 2020-11-12:
哇极其有道理！我服务器和群晖都是老的0.33的上一版了。一语点醒马上试试去。多谢

zyystudio, 2020-11-17:
请问找到zerotier的原因了么？或者你那用的正常？
ps补充换了我服务器上的frp为0.34.1还是一样提示密码不匹配 有啥建议么？实在找不着原因准备卸载安装单独的不要luci界面和zerotier一样的搞下 谢谢

SuLingGG, 2020-11-17:
> 请问找到 zerotier 的原因了么？或者你那用的正常？
> ps 补充换了我服务器上的 frp 为 0.34.1 还是一样提示密码不匹配 有啥建议么？实在找不着原因准备卸载安装单独的不要 luci 界面和 zerotier 一样的搞下 谢谢

在你反馈的第二天我就把 zerotier 的源码切到官方 snapshots 源了，我这边没有使用 zerotier 的条件，你可以试试新版固件，对于你那边 frp 的问题，原因太过繁杂，我无能为力。

closed 2020-11-23 by SuLingGG

zyystudio, 2020-11-24:
供后人参考：
开启zerotier之前先添加个接口
1web界面，网络 -> 接口，添加新接口
名称 ZeroTier
协议选不配置协议，
接口选z开头的适配器。
创建/分配防火墙区域 lan
保存&应用

2.web界面，网络 -> 防火墙-> 通信规则，添加：
名称 Allow-ZeroTier-Inbound
协议 UDP
源区域 任意区域
目标区域 设备
目标端口 9993
保存&应用
即可