SuLingGG/OpenWrt-Rpi #43: Web Admin Settings

truefriend-cz, 2020-11-13:
Hi. I found bug...
If i am removed from "Web Admin Settings" two lines with port :80 and add 0.0.0.0:81 (and for ipv6) then not changes port to 81.
This bug is now, and reboot.

How i can fix this?

SuLingGG, 2020-11-14:
This problem is caused by install offical LuCI by opkg. For details information, plase move to https://github.com/SuLingGG/OpenWrt-Rpi/issues/42

truefriend-cz, 2020-11-14:
No. It is on Original build (2020-11-13, 2020-11-11 tested too)... RPi 4B where Luci is ok.
Clean image (no upgrades, no updates).

SuLingGG, 2020-11-14:
I tested on Raspberrypi 3b, the port is changed correctly.

truefriend-cz, 2020-11-14:
now I'm testing for a Lean build and here the port change works correctly.

In previous cases, the only thing I did with the original image was to change the root password of the user via the "passwd root" command and set the interface for ethernet address assignment from DHCP

SuLingGG, 2020-11-16:
If you are caring about these problems, I think maybe you can try official OpenWrt...

https://downloads.openwrt.org/snapshots/targets/bcm27xx/bcm2711/

RaspberryPi 4 only have snapshots version that isn't include Luci, you should download & flash then install Luci & other modules at SSH.

My firmware is more suitable for Chinese, It includes many Cross-The-GFW packages, these package may be useless for non-Chinese users, If you use official snapshots OpenWrt, you can install packages by your needing. If you find problems, you can make issues to official OpenWrt team, maybe they have ability to solve them.

I am only a compiler that have less ability, most of bugs I can't solve them...

truefriend-cz, 2020-11-16:
Ok. Thanks.
Your poackage is better 👍

closed 2020-11-16 by truefriend-cz