SuLingGG/OpenWrt-Rpi #46: ddns 服务

sklohk, 2020-11-16:
DDNS 服务问题：
在 服务 - 动态 DNS 里的 DDNS 服务供应商 列表为空，只有自定义，对比官方x86的版本 19.07.1或 19.07.3，是在 /etc 下少了 /etc/ddns 目录及目录里的两个文件， services 和 service_ipv6，按 19.07.1 或 3的script，可以自定更新script并append在上述两文件中， 把相应的script 置于 /usr/lib/ddns/下就可以了，但目前 openwrt-rpi的镜像，有些不同，不仅要把update script 置于 /usr/lib/ddns/下， 如 /usr/lib/ddns/update_aliyun_com.sh，还需要在 /usr/share/ddns/default/ 建立对应的文件，如aliyun.com.json，如果没有这个相应 *.json， DDNS 的服务Script就找不到 update script。
修改后就可正常选供应商了，

SuLingGG, 2020-11-16:
[image]
[image]

在我这里 Lean 版 DDNS 工作正常...
如果你是用的是 Offical 版，我必须再次强调一遍，我不对 Offial 版的功能性做出任何保证，除非你有超过一年的 OpenWrt 使用基础，并且具有分析和解决 OpenWrt 问题的能力，否则请使用 Lean 版。

sklohk, 2020-11-16:
了解，我用的是Offical版，开源的应用，都是像你这样的热心人士在努力，我的观点是非商业化产品并不需要对任何版本做任何保证。无论如何，谢谢你的快速回应。

SuLingGG, 2020-11-17:
所以我不太推荐大家使用 Offical 版，Offical 版开发速度较快且激进，很容易在某个版本出现奇奇怪怪的 Bug，而 Lean 版比较保守和稳定，虽然包版本旧了点，但是毕竟设备是拿来用的，而不是用来 fix bugs & kill time 的...
之所以保留 Offical 版是因为收到了不少要求强烈保留 Offical 版的建议。但是目前看来出问题的也往往是 Offical 版居多。

closed 2020-11-23 by SuLingGG