SuLingGG/OpenWrt-Rpi #48: 树莓派最近几个版本的打包文件都不对

swjsj, 2020-11-17:
好不容易下载下来的文件， 计算一下文件的hash值 ，和给你的值完全不一样， 下载文件无法使用

SuLingGG, 2020-11-17:
1. 从你的描述可以看得出来你使用了直接从 Github 下载固件的方式；

2. 直接从 Github 下载文件并不是推荐的下载方式。为什么？见下图：
[image]
为了方便，我贴上这篇文档的链接：
<https://mlapp.cn/1007.html>

3. 由于众所周知的原因，从 Github 下载文件速度堪忧，并且由于线路不稳定，可能有文件损坏的可能；

4. 在文件没有损坏的情况下，你下载到本地的也只是固件的压缩包，解压后才为真实的固件。但是，以树莓派4最新的 Lean 版 Squashfs 格式固件为例：

```
# Acer @ DESKTOP-92AI929 in /c/Users/Acer/Downloads/Compressed/Rpi4-Squashfs [18:38:12]
$ cat sha256sums
4fe8f7981d48c792408b7f8bc14cbe79a4e37059f994b2b4e9f944bec20854b1  openwrt-bcm27xx-bcm2711-rpi-4-ext4-factory.img
348128d1f5c5177a74d70a2c16ae39b42cd406dd00a1fd09c6a597177d9e0dee  openwrt-bcm27xx-bcm2711-rpi-4-ext4-sysupgrade.img
55df454510ea09c571dbd1d09b7fde7d75bf6caac4a152781df50f7ae93b0f29  openwrt-bcm27xx-bcm2711-rpi-4-squashfs-factory.img
83a4b349735eec4a248fa339ab1f740185fd3d36dcf816afba7bca43f67d471c  openwrt-bcm27xx-bcm2711-rpi-4-squashfs-sysupgrade.img

# Acer @ DESKTOP-92AI929 in /c/Users/Acer/Downloads/Compressed/Rpi4-Squashfs [18:38:19]
$ sha256sum openwrt-bcm27xx-bcm2711-rpi-4-squashfs-factory.img
55df454510ea09c571dbd1d09b7fde7d75bf6caac4a152781df50f7ae93b0f29 *openwrt-bcm27xx-bcm2711-rpi-4-squashfs-factory.img

# Acer @ DESKTOP-92AI929 in /c/Users/Acer/Downloads/Compressed/Rpi4-Squashfs [18:38:32]
$ sha256sum openwrt-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.zip
6b8cfab5591f7d026250d9ee17d4a6fa72b29f8197587b686dee08ce039fe6d8 *openwrt-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.zip
```

所以，我提供的 sha256sums 文件中，包含的是对于解压 zip 压缩包镜像的 sha256 校验值，而你如果直接对下载下来的 zip 文件进行校验，当然不会得出正确的结果。

综上所述，可能有以下几种可能性:

1. 由于网络原因，文件下载损坏；

2. 你使用了错误的方式计算固件的校验值；

3. 你没有认真读文档 (根本原因)。

swjsj, 2020-11-17:
感谢,,终于下载好了。 友情提示一下， 文章里面说的点击✅知道下载列表页面，必须要选择一个打包的记录才能看到那个提示区域

`
在“主文章”中点击“✅”即可跳转到对应设备 Artifacts 下载页面。将页面滚动至底部，你可以看到一个 Annotations 提示区域，
`

SuLingGG, 2020-11-17:
> 感谢，，终于下载好了。 友情提示一下， 文章里面说的点击✅知道下载列表页面，必须要选择一个打包的记录才能看到那个提示区域
> 
> `在 “主文章” 中点击 “✅” 即可跳转到对应设备 Artifacts 下载页面。将页面滚动至底部，你可以看到一个 Annotations 提示区域，`

所以我怕你们看不到特意在文档中加了图片呀...
这么贴心、不厌其烦的文档都不看那我真的没什么说的了...

closed 2020-11-17 by swjsj