SuLingGG/OpenWrt-Rpi #50: [重要] 提 Issues 或加群之前，请务必认真阅读文档！

SuLingGG, 2020-11-17:
本项目 Issues 栏目中已存在 **大量** 不认真阅读文档随便开 Issues 进行反馈的条目。

Github Issues 栏目是正经讨论问题的地方，不是懒得看文档直接丢个 Issues 伸手要答案的地方 (加群同理)。

很多普遍的问题我已经写成很详细的文档供大家查阅了，如果你提出与文档所述内容重合度极高的 Issues，在这种情况下我可以理解为是对我劳动成果的漠视，**视情况可能会在不作出任何回应的情况下直接 Close**。

当然，如果你遇到文档中未涉及的问题，或者对项目有什么好的意见和建议，非常欢迎提 Issues 或加群讨论。

所以，为了节约你我的时间，请在提 Issues 或加群前，**务必认真阅读以下文档**：

[**固件特性**](https://mlapp.cn/1001.html): 分类别简要介绍固件内包含的插件和功能

[**功能概览**](https://mlapp.cn/1002.html): 以展开图结构罗列固件内置功能，快速了解固件基本情况

[**版本区别**](https://mlapp.cn/1003.html): 包含 Lean、Offical、Project 三个版本之间的区别

[**设备概览**](https://mlapp.cn/1004.html): 介绍各设备固件的基本情况，同时包含各固件与相关文件的描述与介绍

[**IPV6指南**](https://mlapp.cn/1005.html): 包含借助 `ipv6-helper` CLI 工具快速配置 IPV6 的使用示例

[**命令行工具**](https://mlapp.cn/1006.html): 列举介绍了内置于固件内的命令行工具

[**分区扩容指南**](https://mlapp.cn/1011.html): 介绍了 ext4 和 squashfs 两种格式固件的根分区扩容方法

[**固件下载说明**](https://mlapp.cn/1007.html): 介绍了固件下载的三种方式与固件下载注意事项

[**旁路由设置指南**](https://mlapp.cn/1008.html): 以树莓派为例，介绍了旁路由配置的详细过程

[**网卡支持与推荐**](https://mlapp.cn/1009.html): 介绍了固件对于网卡的支持情况，并对网卡选购给出了一些建议

[**安装及初始设置**](https://mlapp.cn/1010.html): 包含固件在全新安装/升级/以及安装完成后一些初始设置的说明

xsj520, 2021-03-22:
闪迪64g卡不能启动系统，16g能正常启动系统。64g卡刷官方系统能正常进系统。

xsj520, 2021-03-22:
r2s

pcx20010, 2021-04-06:
大佬能编译一次nanopi neo2固件么，软件少点，就出个国。这个设备太适合放车里配合g网卡使用了