SuLingGG/OpenWrt-Rpi #56: 树莓派4B openclash用不了

itnewgay, 2020-12-01:
刷了不同的版本。 但是OPENCLASH总是无法开启。一直提示开启失败。内核版本一直显示未知。
[image]
但是点更新，又会说更新成功。

SuLingGG, 2020-12-02:
首先这不是固件的问题，我的项目确实已经在编译时内置了 Clash 核心，但是核心是通过 OpenClash 项目内下载的:

https://github.com/vernesong/OpenClash/releases/tag/Clash

并且下载的核心也是 OpenClash 开发者经过压缩后的核心 (占用更小 ROM 空间)。

但是经过我的测试，此版 OpenClash 项目内提供的核心在树莓派 4B 上有一些问题，具体表现为在执行二进制文件时，会抛出错误：

`Illegal instruction`

因为在 OpenClash 的 LuCI 里更新内核还是会下载 OpenClash 项目内的内核 (换句话说下载的还是是同一文件)，所以在 OpenClash 的 LuCI 里更新内核并不能解决实质性的问题。

此时可以通过替换 Clash 核心为官方 Clash 项目下未压缩的核心来解决：

进入树莓派 OpenWrt 的 SSH 或者 TTYD 终端，依次逐行执行以下命令即可:

```
cd /etc/openclash/core
rm -rf clash
wget -qO- https://github.com/Dreamacro/clash/releases/download/v1.3.0/clash-linux-armv8-v1.3.0.gz | gunzip -c > clash
chmod +x clash
```

建议在执行以上命令的时候先用 SSR++ 或者 Passwall 在 OpenWrt 上挂一层梯子，然后再执行以上命令；

或者把文件下载到本地，命名为 `clash` 后上传到 `/etc/openclash/core` 目录，最后 chmod +x 赋予可执行权限即可。

itnewgay, 2020-12-02:
感谢回复. 我又单独下载了clash 1.2放上去.可以用了.

closed 2021-03-08 by SuLingGG