SuLingGG/OpenWrt-Rpi #60: 请问如何把22端口ssh的dropbox还原为ttyd的终端？

galaxyskyknight, 2020-12-20:
个人不太喜欢drop box的shell,而且又不用密码认证，暴露给外网不知道怎么弄，有没有办法还原为原生的shell?

SuLingGG, 2020-12-20:
我的 OpenWrt 固件和 dropbox 没啥关系吧，你是不是打错字或者提错项目了呢...？

galaxyskyknight, 2020-12-20:
> 我的 OpenWrt 固件和 dropbox 没啥关系吧，你是不是打错字或者提错项目了呢...？

脑子抽风一下打错字了，是busy box....

galaxyskyknight, 2020-12-20:
还有就是tun0起不来，说错误: Network device is not present, 这个顺便问下，怎么解决或者临时workaround, 我要用openvpn,没tun0不行

SuLingGG, 2020-12-20:
你的描述和实际不符，默认情况下用 SSH 连接到树莓派后命令指示符应该是 zsh 才对，也就是说是这样的输出:

```
# root @ OpenWrt in ~ [16:54:17] 
$ 
```
并且 SSH 连接本身也需要认证，不存在安全性问题。

Web 面板的 TTYD 默认的命令指示符是 busybox 的 ash，TTYD 的 web 服务提供在 7681 端口，是不经登录就能访问的，没办法解决。

你如果想暴露在公网，建议卸载 luci-app-ttyd 或者禁用 ttyd 服务的自启动。

tun0 这个我目前没有解决办法。

galaxyskyknight, 2020-12-20:
我怎么记得以前的老版本ssh 22端口显示的是你说的ash的，且可以设置成root密码登陆的？

SuLingGG, 2020-12-20:
那是很老的版本了。最近的版本我把 SSH 的 Shell 改成了 zsh。
你是不是对 SSH 有什么误解...
SSH (Secure Shell) 不用密码登录那还有安全性可言吗？

galaxyskyknight, 2020-12-20:
> 那是很老的版本了。最近的版本我把 SSH 的 Shell 改成了 zsh。
> 你是不是对 SSH 有什么误解...
> SSH (Secure Shell) 不用密码登录那还有安全性可言吗？

好吧，我刚发现是我的term里面存了密码，所以.....-_-!,   tun0是什么情况呀。。。要等上游修复么？

SuLingGG, 2020-12-20:
我觉得它只是显示上的问题。设备已经分配给 tun0 了，但是可能是因为 tun0 未工作所以显示这个错误，忽略即可。启用 OpenVPN 服务器就不会出现这个错误了：
如图所示：
[image]
[image]

closed 2020-12-22 by SuLingGG