SuLingGG/OpenWrt-Rpi #62: 固件会core掉，从系统up时间看出来的，半夜里重启了，但没开任何自动重启任务，请问如何调试？

galaxyskyknight, 2020-12-21:
系统半夜里core掉了（推算是4点多），1220的x86 squashfs版本。有coredump可以看吗？很频繁，一天好几次，看概览-状态-系统里的up时间有中断。有时候不知道怎么突然一下就core掉了，硬件没问题，昨天才从19.x(4.x内核）升上来的，之前连续开机50天无宕机。core掉了内核日志和系统日志都被刷没了，也不知道是哪个进程引起的。
[image]

SuLingGG, 2020-12-21:
恢复初始设置试试看。
如果你用的是 ESXI 的话，可能对 5.x 内核或者 squashfs 格式的固件不太友好，可能会出现重启的情况。

galaxyskyknight, 2020-12-21:
没用EXSI,直接用diskimage干净刷的, 刷完都是默认出厂设置，然后全手工做的各项插件和系统设置，应该都是干净的。这个没办法生产core文件，调试trace coredump么？

SuLingGG, 2020-12-21:
/sys/kernel/debug/crashlog
如果这里没有的话那就没有办法了...

galaxyskyknight, 2020-12-21:
> /sys/kernel/debug/crashlog
> 如果这里没有的话那就没有办法了...

这个目录下，crash时间4:16产生的文件倒是有一堆，但是就是没有crashlog, -_-!!
[image]

closed 2020-12-22 by SuLingGG