SuLingGG/OpenWrt-Rpi #64: 无法更新 xray和v2ray

6668412, 2020-12-23:
更新xray时报错:
您没有填写 Xray 路径。请保存应用后再手动更新。
但下面有默认路径   （重新复制粘贴一下路径，已解决，可能是个小bug？）
更新v2ray时报错:
无法移动新文件到：/usr/bin/v2ray/

SuLingGG, 2020-12-23:
我这儿好像没有这个问题...
[image]
[image]
[image]

6668412, 2020-12-23:
我重新下载并安装了固件，但是还是出现这个问题，并且又发现无法保存更改的配置（右上角提示）
[image]

6668412, 2020-12-23:
[image]

SuLingGG, 2020-12-23:
先把这个文件刷入 SD 卡，无需插卡开机，紧接着再刷入我的固件试一试。
[Restore-SDCard-2G.img.zip](https://github.com/SuLingGG/OpenWrt-Rpi/files/5735129/Restore-SDCard-2G.img.zip)

6668412, 2020-12-23:
我试了一下，固件更新貌似好了，但是不能保存配置的问题还在
[image]

SuLingGG, 2020-12-23:
那说明你这张 SD 卡生命快走到尽头了，换张卡吧...

6668412, 2020-12-23:
哈哈，谢谢了，我会为这张卡寻找一个合适的归宿的^^

closed 2020-12-23 by 6668412