SuLingGG/OpenWrt-Rpi #76: 搜不到树莓派发出的WiFi信号

hmsjy2017, 2021-02-02:
搜不到树莓派发出的WiFi信号，树莓派上无线强度0%
[image]

SuLingGG, 2021-02-02:
我这里专门测试了一下，最新版固件没有这个问题。

[image]

OpenWrt 上的无线强度是回程信号，你可以理解为是连接 WiFi 的设备感受到的无线信号。
再换句话说就是手机状态栏上面的 WiFi 信号强度。
OpenWrt 显示的信号强弱与 OpenWrt 自身无线工作是否正常没有必然的关系，如果没有设备连接，那么 OpenWrt 上面的信号强度就是 0。
另外建议尽量不要给树莓派的 WiFi 上密码，如果确有安全需求，建议直接隐藏 SSID，如果给 WiFi 上密码有可能会触发一些玄学问题。
建议你按这里的教程:

https://openwrt.cc/restore

操作一遍，刷入恢复固件后紧接着刷入 OpenWrt 固件，不要动与无线相关的设置，也不要恢复原来旧固件的备份文件重新试一试。

hmsjy2017, 2021-02-02:
我怀疑是电脑网卡不支持5G WiFi，我家路由器也只能搜到2.4G
OpenWrt默认频段好像是5G，如何改为2.4G？

SuLingGG, 2021-02-02:
那你这前后文明显有问题啊。
你说 1 月 28 号的固件没有这个问题，但是我的固件默认一直都是 5G 频段。或者说你之前和现在用的设备不一致？
另外我还是建议你用支持 5Ghz 频段的设备试试，现在的手机一般都支持 5Ghz 了，除非是四五年前的手机...
这是设置 2.4Ghz 的方法:
[image]

hmsjy2017, 2021-02-02:
之前和现在用的设备不一致，电脑仍然搜不到信号....忘说了
iPad 能搜到

SuLingGG, 2021-02-02:
> 之前和现在用的设备不一致，电脑仍然搜不到信号.... 忘说了
> iPad 能搜到

那改成 2.4Ghz 应该没问题了。但是最高无线速率会降低，最高只有 72Mbit/s。

hmsjy2017, 2021-02-02:
已解决

closed 2021-02-02 by hmsjy2017