SuLingGG/OpenWrt-Rpi #78: 作为主路由后，下联的所有终端NTP会失败

znbcr, 2021-02-02:
非常感谢大佬们编译这个FW

但是在用它替换掉家中的主路由器后，发现一个问题：下联网络的所有设备 从NTP server获得校准时钟都会失败

我手里有Raspberry 2B+ 和 3B+ ，目前测试都出现了这个问题。网络ISP为中国移动光纤宽带，下联的终端有Windows 10系统的电脑，也有 Android系统的TV box，全部都不能正确校准时间。目前我用的都是ext 的img，正在准备换sq的试试看

SuLingGG, 2021-02-04:
我这边没有这个问题...

并且在一年多内也没有收到有关这个问题的反馈。

如果你手头有安卓设备，可以尝试使用这个 APP:

https://www.apkmirror.com/apk/sergey-baranov/clocksync-2/

测试一下是否可以正常同步时间。

如果默认的 NTP 服务器不可用，可以尝试更换阿里云提供的公网 NTP 服务器试一试...

aka714, 2021-02-06:
我也遇到同样的问题，树莓派4B做旁路由（自身IP192.168.1.45），固件版本OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)，树莓派4B的网关及DNS指向主路由（192.168.1.1），网络内其余设备网关和DNS均指向树莓派4B（192.168.1.45）。

出现问题：
网络内的全部设备一切网络行为都正常，就是时间怎么也不能同步。

初步尝试：
更换各种外部NTP服务器，均能ping得通它们，但是就是不能同步时间。
无论设置什么NTP都没用，什么阿里云的、腾讯的、微软的。。。。。全部无法同步，但是在那个设备上都能ping通这个NTP服务器。

解决步骤：
1）在树莓派4B管理页面【系统】-【系统】-【时间同步】页面勾上“启用 NTP 客户端”和“作为 NTP 服务器提供服务”；
2）网络里需要时间同步的设备的NTP设置均指向树莓派4B（192.168.1.45），相当于用树莓派转一手，就别折腾使用外部的NTP服务器了；
3）搞定，所有设备时间同步成功。

nemo2018, 2021-02-20:
也遇到过这个问题，简单，就把openwrt做NTP server就行了，快速稳定。

harry-he-sw, 2021-05-06:
打开防火墙 123 端口

baysonfox, 2021-05-30:
同，自编译的固件无问题，更换固件后出现，目前使用@aka714提供的解决方案解决.