SuLingGG/OpenWrt-Rpi #79: Luci  Bad Gateway

hmsjy2017, 2021-02-03:
执行
```
opkg list-upgradable | cut -f 1 -d ' ' | xargs opkg upgrade
```
命令后浏览器打开 Luci 显示
```
Bad Gateway
The process did not produce any response
```
docker 无法启动容器
ssh 能连上

SuLingGG, 2021-02-03:
这个问题很复杂。

OpenWrt 和 CentOS，Ubuntu 这些发行版不一样，是针对嵌入式设备的，而不是生产环境。
这也意味着 OpenWrt 没有必要对包版本有过高的期望。

本项目 Lean 版固件基于 

<https://github.com/coolsnowwolf/lede>

是 Lean 基于官方 OpenWrt 进行修改/本土化的 OpenWrt 项目，其中的源码和官方源码相差很大，而固件中内置的 opkg 软件源是适用于 OpenWrt snapshot 和 LuCI 19.07 的，换句话说就相当于用魔改版源码编译出来的固件却用了官方源。

在这种情况下，安装一些无关紧要的包比如 vim，screen 之类的一般不会出问题，但是如果用 opkg 安装稍微接近系统核心一点的软件包大概率会出现 Bug，甚至导致固件炸掉。

而网络上流传的 “升级 OpenWrt 所有软件包” 的方法是基于 OpenWrt 官方固件的。官方固件用官方源升级自然问题比魔改版少，但是实际上，即使你用官方版固件，用这个命令升级所有包大概率也会出现一些问题。

所以你能在恩山之类的地方能经常看到，某固件作者的帖子里经常会说，刷入自己的固件建议不保留配置升级。

实际上，即使同样是我的固件，刷入新版本固件后，如果你要还原旧版本的备份归档文件，可能也有小概率出现各种各样的问题。

同一作者同一源码在新旧固件升级是已然如此，用官方版软件源升级魔改版固件怎么可能不出问题呢？

所以反观 OpenWrt 里面的 opkg，没有直接提供 upgrade 的参数不是没有原因的。

hmsjy2017, 2021-02-03:
只能重新刷固件吗？

SuLingGG, 2021-02-03:
> 只能重新刷固件吗？

只能重刷固件。

hmsjy2017, 2021-02-03:
如果用的是 squashfs 格式的固件，能通过 SSH 系统还原吗？（虽然我用的是ext4固件...）
文档里说是可以的，但没有讲具体方法。
[image]

SuLingGG, 2021-02-03:
按理说如果是 squashfs 固件是可以还原的。但是你如果通过这种方式升级后想要还原，我不做保证。

hmsjy2017, 2021-02-03:
如果要还原应该怎么操作呢？

SuLingGG, 2021-02-04:
> 如果要还原应该怎么操作呢？

建议百度一下...这个问题到处都能百度到。

closed 2021-02-05 by hmsjy2017