SuLingGG/OpenWrt-Rpi #83: 每天会kernel crash导致自动重启

cxfksword, 2021-02-09:
版本如下：

```
$ uname -a
Linux OpenWrt 5.4.92 #0 SMP Mon Feb 1 03:22:35 2021 x86_64 GNU/Linux
```

crash日志：

```
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.767970]  nf_hook_slow+0x42/0xc0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.771749]  ip_rcv+0x6e/0xa0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.775002]  ? ip_rcv_finish_core.isra.18+0x2d0/0x2d0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.596791] general protection fault: 0000 [#1] SMP PTI
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.780369]  __netif_receive_skb_one_core+0x63/0x70
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.785568]  __netif_receive_skb+0x1e/0x60
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.602568] CPU: 1 PID: 10114 Comm: transmission-da Tainted: G         C        5.4.92 #0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.789944]  netif_receive_skb+0x28/0x80
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.794161]  ? fdb_find_rcu+0xce/0x120
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.798210]  br_netif_receive_skb+0x2f/0x40
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.802706]  br_pass_frame_up+0x10b/0x140
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.611166] Hardware name: Default string Default string/Default string, BIOS 5.12 04/07/2019
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.807013]  br_handle_frame_finish+0x1af/0x450
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.811854]  ? br_pass_frame_up+0x140/0x140
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.620095] RIP: 0010:native_queued_spin_lock_slowpath+0x13e/0x1c0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.626616] Code: 02 89 c1 c1 e1 10 0f 84 8d 00 00 00 c1 e9 12 83 e0 03 83 e9 01 48 c1 e0 05 48 63 c9 48 05 c0 03 02 00 48 03 04 cd 80 26 09 82 <48> 89 10 8b 42 08 85 c0 75 09 f3 90 8b 42 08 85 c0 74 f7 48 8b 02
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.816322]  br_nf_hook_thresh+0xb3/0x12a0 [br_netfilter]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.822050]  ? br_pass_frame_up+0x140/0x140
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.826542]  br_nf_hook_thresh+0xa3b/0x12a0 [br_netfilter]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.832360]  ? br_pass_frame_up+0x140/0x140
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.646024] RSP: 0000:ffffc9000003c6f0 EFLAGS: 00010286
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.836843]  ? ipv4_cow_metrics+0x10/0x10
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.841154]  ? 0xffffffffa18f52df
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.651563] RAX: ff2e9754000203c0 RBX: ffff88815ad80280 RCX: 000000000000045b
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.844756]  ? nf_hook_slow+0x42/0xc0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.848718]  br_nf_hook_thresh+0xe54/0x12a0 [br_netfilter]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.854529]  ? br_nf_hook_thresh+0x910/0x12a0 [br_netfilter]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.860530]  br_handle_frame+0x17c/0x370
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.659086] RDX: ffff88816ad203c0 RSI: 0000000000080000 RDI: ffff88815ad80284
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.666608] RBP: ffffc9000003c6f0 R08: 0000000000000000 R09: 00000000cb02a8c0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.864750]  ? br_pass_frame_up+0x140/0x140
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.869243]  ? br_handle_frame_finish+0x450/0x450
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.674134] R10: 0000000000000001 R11: 000000000000eab3 R12: ffff88815ad80284
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.874261]  __netif_receive_skb_core+0x1fb/0xb40
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.879279]  __netif_receive_skb_one_core+0x2a/0x70
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.884472]  __netif_receive_skb+0x1e/0x60
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.888853]  process_backlog+0xb0/0x170
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.681621] R13: ffffc9000003c740 R14: 0000000000000001 R15: ffff888160c08fc0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.892985]  __napi_poll+0x2b/0x90
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.896658]  net_rx_action+0xca/0x210
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.689116] FS:  00007fc42d86eb20(0000) GS:ffff88816ad00000(0000) knlGS:0000000000000000
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.697624] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.900622]  __do_softirq+0xc9/0x20e
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.904489]  irq_exit+0x6f/0x80
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.703694] CR2: 00007f5ed33e0090 CR3: 000000011cf38006 CR4: 00000000003606e0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.907898]  smp_call_function_interrupt+0x38/0x40
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.912992]  smp_call_function_single_interrupt+0x9/0x10
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.918628]  call_function_single_interrupt+0xf/0x20
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.923895]  </IRQ>
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.711233] Call Trace:
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.713934]  <IRQ>
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.926258] RIP: 0033:0x7fc42dc51866
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.930113] Code: 89 c0 72 63 bf 01 00 00 00 74 27 89 f7 4c 8b 0c fa 8d 7e ff 48 8b 3c fa 4c 8b 5f 70 4d 39 59 70 75 08 48 8b 7f 78 49 39 79 78 <40> 0f 9f c7 40 0f b6 ff 29 fe 89 f7 48 8b 34 f2 4c 8b 56 70 4c 39
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.949528] RSP: 002b:00007fc42d86e9c0 EFLAGS: 00000297 ORIG_RAX: ffffffffffffff04
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.957493] RAX: 0000000000000001 RBX: 0000000002435e80 RCX: 00007fc42d913340
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.965028] RDX: 0000000002471500 RSI: 0000000000000004 RDI: 0000000002251a60
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.972551] RBP: 00007fc42dc34b00 R08: 0000000000000001 R09: 00007fc42dbf1e00
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.716190]  _raw_spin_lock+0x1a/0x20
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.720131]  fast_classifier_recv+0x1d23/0x1f68 [fast_classifier]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.980066] R10: 0000000002471500 R11: 000000000001efac R12: 0000000000066f9c
[... 7 lines cut ...]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.748612]  ip_forward_finish+0x3a/0x50
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126895.085009]  gspca_spca561 gspca_spca508 gspca_spca506 gspca_spca505 gspca_spca501 gspca_spca500 gspca_sonixj gspca_sonixb gspca_sn9c20x gspca_se401 gspca_pac7311 gspca_pac207 gspca_ov534_9 gspca_ov534 gspca_ov519 gspca_mr97310a gspca_mars gspca_main dm9601 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a ath9k_pci_owl_loader asix amdgpu xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_esp xt_ecn xt_dscp xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY xr_usb_serial_common wlcore_sdio wireguard videobuf2_v4l2 videobuf2_common via_velocity via_rhine usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm solos_pci snd_usbmidi_lib snd_intel_nhlt snd_hda_core slhc sky2 skge sis900 sis190 sch_cake rtl8150 regmap_core radeon r8712u(C)
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.752822]  ip_forward+0x391/0x430
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126894.756582]  ? ip4_obj_hashfn+0xb0/0xb0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126895.174313]  r8169 r8168 r8152 r8125 r6040 pegasus pcnet32 niu nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4 ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core macvlan lzo lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 kvm_intel kvm kaweth irqbypass iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables igc i40e hso gpu_sched forcedeth exfat ethoc et131x e1000e e100 dmx_usb crc8 [... cut ...]
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126895.600273] FS:  00007fc42d86eb20(0000) GS:ffff88816ad00000(0000) knlGS:0000000000000000
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126895.608770] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126895.614834] CR2: 00007f5ed33e0090 CR3: 000000011cf38006 CR4: 00000000003606e0
2021-02-09T10:23:02+08:00 OpenWrt kernel: [126895.622368] Kernel panic - not syncing: Fatal exception in interrupt
```

只看到熟悉的transmission字样，其他看不明

SuLingGG, 2021-02-09:
在 "网络" - "Turbo ACC 网络加速" 里关掉 SFE 加速试试。

cxfksword, 2021-02-11:
我关了后，近一天是没重启了，我再观察下

leungzw, 2021-02-15:
同样问题，之前每天不定时重启两次。
关闭后稳定运行3天了

bg4drl, 2021-02-15:
CPU高值基本都是60%超 还是等源码后再看GG更新

closed 2021-03-08 by SuLingGG

sanqudui8ban, 2021-08-25:
请问你是怎么在kernel版本5.4下打开内核崩溃日志记录的