SuLingGG/OpenWrt-Rpi #84: IPv6应该怎么配置啊?

orangejx, 2021-02-10:
IPv6应该怎么配置啊?
设备是Raspberry Pi 4 Model B Rev 1.1
上级网关设备是上海电信的SDN网关, 现在是只能获取到/64地址, 获取不到pd
IPv6-helper的三种模式 server relay hybrid都用不了, 表现情况是Lan口获取不到v6地址, WAN6可以通过DHCPv6获取到v6 /64地址. LAN口在设置了全局ULA前缀/64之后依然获取不到设置的全局ULA pd

orangejx, 2021-02-10:
emmmm 混合模式的hybrid是不是手误打错了呀,

orangejx, 2021-02-10:
/允悲 终于找到原因了.
1. 替换内容,要将以下内容替换掉.
   1.1 内容|替换内容|文件位置
   ---|:--:|---:
   hybird|hybrid|`/bin/ipv6-helper` (`/scripts/ipv6-helper.sh`)
   `WAN`|`wan`|`/etc/config/network`
   `option ifname ... `|`option ifname @wan`|`/etc/config/network` 中的 `config interface 'wan6'`内的option ifname
2. 重装`ipv6-helper`
   2.1 `ipv6-helper remove`
   2.2 `ipv6-helper install`

在替换`hybird`为`hybrid`时, 使用vi批量替换会快一些. `:%s/hybird/hybrid/g`

caibok80, 2021-02-17:
好像有个BUG，你的WAN6如果没有”DHCPPV6客户端“模式就要执行“ipv6-helper install“然后正常设置就好了。

orangejx, 2021-02-18:
> 好像有个BUG，你的WAN6如果没有”DHCPPV6客户端“模式就要执行“ipv6-helper install“然后正常设置就好了。

默认是没安装ipv6模块的，所以如果要使用wan6，是需要安装ipv6模块的[image]

closed 2021-03-08 by SuLingGG