SuLingGG/OpenWrt-Rpi #87: 树莓派4B，大家有没有遇到管理界面经常打不开的情况

kfcceo, 2021-02-18:
大家有没有遇到管理界面经常打不开的情况，需要重启才能打开。

xukongyang, 2021-02-19:
有的，有的，我的是树莓派4b的4g版，刷的是今天的sysquad最新版本，今天碰到很多次管理界面打不开（显示一片空白），要过好久（大概个把小时）才能再次显示管理界面，然后才能正常操作路由器（我倒是没有尝试界面一打不开就重启）。

SuLingGG, 2021-02-19:
建议换张好点的 SD 卡试试。

xukongyang, 2021-02-19:
> 建议换张好点的 SD 卡试试。

谢谢答复，用的是雷克萨的667x 128g的sd卡了，质量应该还行的，在网上找到个帖子 https://github.com/jerrykuku/luci-theme-argon/issues/27 ，按里面的方法

能ssh么？能的话 进ssh 修改 vi /etc/config/luci 修改 option mediaurlbase '/luci-static/argon' 这个选项为 option mediaurlbase '/luci-static/bootstrap' 然后保存 重启

试了，重启后倒是能访问管理界面了，不知道是解决了还是重启大法暂时接解决了，我再观察下试试。。。

kfcceo, 2021-02-19:
> > 建议换张好点的 SD 卡试试。
> 
> 谢谢答复，用的是雷克萨的667x 128g的sd卡了，质量应该还行的，在网上找到个帖子 [jerrykuku/luci-theme-argon#27](https://github.com/jerrykuku/luci-theme-argon/issues/27) ，按里面的方法
> 
> 能ssh么？能的话 进ssh 修改 vi /etc/config/luci 修改 option mediaurlbase '/luci-static/argon' 这个选项为 option mediaurlbase '/luci-static/bootstrap' 然后保存 重启
> 
> 试了，重启后倒是能访问管理界面了，不知道是解决了还是重启大法暂时解决了，我再观察下试试。。。

closed 2021-02-19 by kfcceo