SuLingGG/OpenWrt-Rpi #119: 建议取消编译树莓派3B无线

zuozhehao, 2021-04-08:
建议取消编译树莓派3B无线。
使用3B当旁路由，关闭无线，每隔一段时间就会自动重启网卡。

SuLingGG, 2021-04-08:
@zuozhehao 可以看看是不是供电或者接触问题。
我这里也有 3b，没有出现过这个问题。
另外无线主要是供初次设置使用，去掉无线会很不方便，所以暂时不会考虑...

zuozhehao, 2021-04-08:
> @zuozhehao 可以看看是不是供电或者接触问题。
> 我这里也有 3b，没有出现过这个问题。
> 另外无线主要是供初次设置使用，去掉无线会很不方便，所以暂时不会考虑...

用的3A的电源适配器，
```
Wed Mar 24 01:19:00 2021 kern.warn kernel: [ 9885.926451] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected
Wed Mar 24 01:19:00 2021 kern.warn kernel: [ 9885.926451]
Wed Mar 24 01:19:00 2021 kern.warn kernel: [ 9885.939240] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected
Wed Mar 24 01:19:00 2021 kern.warn kernel: [ 9885.939240]
Wed Mar 24 01:19:00 2021 kern.info kernel: [ 9885.949162] usb 1-1: USB disconnect, device number 2
Wed Mar 24 01:19:00 2021 kern.info kernel: [ 9885.959592] usb 1-1.1: USB disconnect, device number 3
Wed Mar 24 01:19:00 2021 kern.info kernel: [ 9885.967787] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Wed Mar 24 01:19:00 2021 daemon.notice netifd: Network device 'eth0' link is down
Wed Mar 24 01:19:00 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Wed Mar 24 01:19:00 2021 kern.info kernel: [ 9885.983544] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Wed Mar 24 01:19:00 2021 daemon.notice netifd: Interface 'lan' is now down
Wed Mar 24 01:19:00 2021 daemon.notice netifd: Interface 'lan' is disabled
Wed Mar 24 01:19:01 2021 kern.warn kernel: [ 9886.433054] Indeed it is in host mode hprt0 = 00021501
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9886.628998] usb 1-1: new high-speed USB device number 4 using dwc_otg
Wed Mar 24 01:19:01 2021 kern.warn kernel: [ 9886.638464] Indeed it is in host mode hprt0 = 00001101
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9886.853276] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9886.866976] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9886.877598] hub 1-1:1.0: USB hub found
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9886.884262] hub 1-1:1.0: 5 ports detected
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9887.181050] usb 1-1.1: new high-speed USB device number 5 using dwc_otg
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9887.289373] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00, bcdDevice= 2.00
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9887.303321] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Wed Mar 24 01:19:01 2021 kern.info kernel: [ 9887.316321] smsc95xx v1.0.6
Wed Mar 24 01:19:02 2021 kern.info kernel: [ 9887.391623] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:46:d2:8b
Wed Mar 24 01:19:02 2021 daemon.notice netifd: Interface 'lan' is enabled
Wed Mar 24 01:19:02 2021 daemon.notice netifd: Interface 'lan' is setting up now
Wed Mar 24 01:19:02 2021 daemon.notice netifd: Interface 'lan' is now up
Wed Mar 24 01:19:02 2021 kern.info kernel: [ 9887.491437] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Wed Mar 24 01:19:02 2021 user.notice firewall: Reloading firewall due to ifup of lan (eth0)
Wed Mar 24 01:19:03 2021 kern.info kernel: [ 9889.087049] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Wed Mar 24 01:19:03 2021 daemon.notice netifd: Network device 'eth0' link is up
Wed Mar 24 01:19:03 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Wed Mar 24 01:19:03 2021 kern.info kernel: [ 9889.097912] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xCDE1
Wed Mar 24 04:00:00 2021 cron.err crond[866]: USER root pid 15826 cmd /etc/coremark.sh
Wed Mar 24 10:23:54 2021 kern.warn kernel: [42579.880528] ERROR::dwc_otg_hcd_urb_enqueue:501: Not connected
Wed Mar 24 10:23:54 2021 kern.warn kernel: [42579.880528]
Wed Mar 24 10:23:54 2021 kern.info kernel: [42579.939346] usb 1-1: USB disconnect, device number 4
Wed Mar 24 10:23:54 2021 kern.info kernel: [42579.947320] usb 1-1.1: USB disconnect, device number 5
Wed Mar 24 10:23:54 2021 kern.info kernel: [42579.955670] smsc95xx 1-1.1:1.0 eth0: unregister 'smsc95xx' usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet
Wed Mar 24 10:23:54 2021 kern.info kernel: [42579.971400] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
Wed Mar 24 10:23:54 2021 daemon.notice netifd: Network device 'eth0' link is down
Wed Mar 24 10:23:54 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Wed Mar 24 10:23:54 2021 daemon.notice netifd: Interface 'lan' is now down
Wed Mar 24 10:23:54 2021 daemon.notice netifd: Interface 'lan' is disabled
Wed Mar 24 10:23:54 2021 kern.warn kernel: [42580.443193] Indeed it is in host mode hprt0 = 00021501
Wed Mar 24 10:23:55 2021 kern.info kernel: [42580.639155] usb 1-1: new high-speed USB device number 6 using dwc_otg
Wed Mar 24 10:23:55 2021 kern.warn kernel: [42580.648702] Indeed it is in host mode hprt0 = 00001101
Wed Mar 24 10:23:55 2021 kern.info kernel: [42580.863443] usb 1-1: New USB device found, idVendor=0424, idProduct=9514, bcdDevice= 2.00
Wed Mar 24 10:23:55 2021 kern.info kernel: [42580.877683] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Wed Mar 24 10:23:55 2021 kern.info kernel: [42580.888546] hub 1-1:1.0: USB hub found
Wed Mar 24 10:23:55 2021 kern.info kernel: [42580.895440] hub 1-1:1.0: 5 ports detected
Wed Mar 24 10:23:55 2021 kern.info kernel: [42581.191238] usb 1-1.1: new high-speed USB device number 7 using dwc_otg
[... cut ...]
Wed Mar 24 10:23:55 2021 kern.info kernel: [42581.317737] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
Wed Mar 24 10:23:55 2021 kern.info kernel: [42581.330921] smsc95xx v1.0.6
Wed Mar 24 10:23:55 2021 kern.info kernel: [42581.400012] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:46:d2:8b
Wed Mar 24 10:23:55 2021 daemon.notice netifd: Interface 'lan' is enabled
Wed Mar 24 10:23:55 2021 daemon.notice netifd: Interface 'lan' is setting up now
Wed Mar 24 10:23:55 2021 daemon.notice netifd: Interface 'lan' is now up
```

SuLingGG, 2021-04-08:
日志不一定能看出来。
你可以先换根线试试，当然也不排除是板子的体质问题...

Gestabo, 2021-04-08:
建议换2.1A以上的品牌充电头，有挂盘的话3A以上。我的树莓派3B一直稳定

FWangZil, 2021-04-09:
额，你的 3B 不稳定就让人停止编译吗......这个回路真是绝了。

closed 2021-04-10 by zuozhehao