SuLingGG/OpenWrt-Rpi #125: SuLing GG你好，能优化下树莓派4b对无线网卡RTL8812b的支持吗（华硕USB-AC57）

yangwen9, 2021-04-14:
SuLing GG你好，能优化下树莓派4b对无线网卡RTL8812b的支持吗（华硕USB-AC57），目前该网卡没法被树莓派4b识别&兼容，但是这个华硕USB-AC57网卡能被Nano pi R2S很友好的识别和兼容（Klever 1988的固件），我目前200M的宽带，这个无线网卡能装在R2S上能跑满带宽，树莓派自身的WIFI信号太鸡肋了，所以一直在想外接一个信号强且能跑满200M带宽的无线网卡，华硕USB-AC57目前信号和无线速度这两方面都觉得不错（其他型号网卡COMFAST 726B也试过了，它能被树莓派4b驱动，但是断电或者重启后，无线网卡WIFI就自动关掉或者系统就不识别了，需要重新配置或者重启才能重新启动，另外COMFAST 72B的5G无线WIFI连接速度只能达到50M左右），恳请SuLingGG能针对这个issue进行下优化，这样树莓派4b OpenWrt就完美了！！！

SuLingGG, 2021-04-14:
我在文档里说过：珍爱生命，远离螃蟹卡。
同时我文档也提到：“我不会对 ‘某外接设备’不可用” 负责”。
我没办法保证我的固件对于每一款无线网卡都有良好的支持，网卡芯片型号千差万别，我这里不可能拥有所有设备，也没办法对某款设备进行测试。
我这里只能确定我的固件对以下几个芯片支持得不错:
RT3070
MT7612U
如果要让我选出一款支持最完美的，那就是网件 A6210。闲鱼 100 以下能买到。
建议使用且仅使用我文档里支持的网卡。

yangwen9, 2021-04-14:
谢谢GG建议和反馈，网件A6210在我购物车有段时间了，没买的原因主要是体积有点大，听了您的建议，我准备还是淘一个NETGEAR A6210试试，性能和兼容OK是关键，非常感谢！

EnnawYang, 2021-04-14:
COMFAST 726B（RTL8821CU）驱动与树莓派板载无线驱动冲突，禁用板载wifi可解决，目前无有效解决方案

yangwen9, 2021-04-14:
我说呢，特别是树莓派4b重启或者断电再开机后，USB无线网卡信号就失效，要在无线设置界面里切换COMFAST 726B的其他无线频段或者其他信号频率才能重新生效

AmadeusGhost, 2021-04-15:
rtl8812bu 的驱动用的是 https://www.github.com/immortalwrt/immortalwrt/tree/master/package/kernel/rtl88x2bu ，你可以自己加进去试试。

yangwen9, 2021-04-15:
谢谢您的建议和帮助，刚接触Openwrt几个月， 我不会编译OpenWrt系统， 正在努力接触和学习中， 安装rt8812bu这个驱动具体怎么操作？ 像安装其他插件一样，先上传，然后再安装？ 
或者麻烦SuLing GG在下个版本更新Release前把rt8812bu这个驱动给加进树莓派4b上去？ （https://www.github.com/immortalwrt/immortalwrt/tree/master/package/kernel/rtl88x2bu） 后面我刷新固件测试后，反馈测试结果和兼容性？ 感谢！

SuLingGG, 2021-04-18:
驱动已尝试加入，可以下载最近编译的固件试试看。

yangwen9, 2021-04-18:
好的，谢谢，网件A6210 的无线网卡今天也会到货,到时候华硕和网件两个网卡一起试试

yangwen9, 2021-04-18:
华硕USB-AC57无线网卡今天下午刚用最新的树莓派4b固件调试了下，能很好的兼容，重启或者断电后，重新开机，无线网卡的信号能自动调出来，后续就看稳定不稳定了，使用一段时间后再反馈。 
网件A6210的货还没到，到货后再调试，再次感谢Suling GG 大佬

yangwen9, 2021-04-18:
今天下午还发现了一个问题，4b的板载WIFI信道设定的默认值是36信道，如果想修改成其他信道，例如149信道，然后点“保存&应用”，在状态栏里显示的信道还是36，后面应该是起冲突了，WIFI信号就没有了，只能通过有线连到树莓派4b后台去重新修改成36信道（吸取了教训，买了一个USB有线网卡，以防WIFI出问题，无法进入后台），然后重启，才正常，简单点，目前的问题是对于树莓派4b自身的板载wifi，修改了信道后，貌似系统没有收到反馈，状态栏里显示的还是36信道，与设定的149信道不一致了，所以WIFI就无法连接了

[image]

EnnawYang, 2021-04-18:
目前源码中的 `cypress-firmware` 驱动可以使用36-136信道

yangwen9, 2021-04-18:
喔，这样啊，建议在选项里面把136信道以上的给删除掉，避免只要选一个136以上的信道，WIFI就直接瘫痪了，只能通过有线进后台，或者技术好的，修改sd卡里面个配置文件进后台

EnnawYang, 2021-04-18:
[image]

[cypress-firmware-43455-sdio_v5.4.18-2020_0925-3_aarch64_cortex-a72.zip](https://github.com/SuLingGG/OpenWrt-Rpi/files/6330854/cypress-firmware-43455-sdio_v5.4.18-2020_0925-3_aarch64_cortex-a72.zip)

你可以试下这个版本的驱动，支持36-161信道，解压后安装ipk

yangwen9, 2021-04-18:
好的，我试试，还有一个问题，网件A6210的WiFi可以正常驱动使用，但是无线网卡的工作指示灯不亮，这个通过优化固件解决吗？华硕无线网卡工作时的指示灯是亮的

yangwen9, 2021-04-18:
cypress-firmware-43455-sdio_v5.4.18-2020_0925-3_aarch64_cortex-a72.zip 这个驱动很好使，装上重启后，149后面信道可以使用了，用了这个信道后WIFI无线信号也提升到200M上了，并且树莓派4b Raspberry pi OS系统可以识别这个WIFI信号了（之前是不能识别36信道的5G信号的），@SuLing GG建议下个版本的Release，把EnnawYang推荐的这个驱动打包进去，谢谢了！

EnnawYang, 2021-04-18:
> cypress-firmware-43455-sdio_v5.4.18-2020_0925-3_aarch64_cortex-a72.zip 这个驱动很好使，装上重启后，149后面信道可以使用了，用了这个信道后WIFI无线信号也提升到200M上了，并且树莓派4b Raspberry pi OS系统可以识别这个WIFI信号了（之前是不能识别36信道的5G信号的），@suling GG建议下个版本的Release，把EnnawYang推荐的这个驱动打包进去，谢谢了！

我测试了下网速好像没那么快吧，你有实测过吗

yangwen9, 2021-04-18:
我待会再实测下，截个图片，有时候不稳定，高的时候快200

yangwen9, 2021-04-18:
我刚又实测了一下，下载140M，上传95M，我已经很满足了，现在的唯一问题就是网件这款无线网卡的指示灯没法被点亮
[image]

yangwen9, 2021-04-21:
@SuLingGG 你好， 最新的4/20号的树莓派4b固件，无线信号的信道设定还是有点bug, 如果我选择的信道是153，状态里面显示的是149，如果选择的是161，状态里面显示的是157，以此类推
[image]

yangwen9, 2021-04-22:
@SuLingGG @EnnawYang 大佬你好，上面反馈的这个信道不一致问题，在最近的固件Release里面有更新进去fix这个问题吗？如果有更新进去的话，我今天再下个固件试试，谢谢了！

EnnawYang, 2021-04-22:
> @SuLingGG @EnnawYang 大佬你好，上面反馈的这个信道不一致问题，在最近的固件Release里面有更新进去fix这个问题吗？如果有更新进去的话，我今天再下个固件试试，谢谢了！

完全没有头绪，另外我只是个Vegetable Chicken

yangwen9, 2021-04-26:
最新的固件貌似把这个问题给解决了，华硕和网件的无线网卡都能正常显示设定的信道了

closed 2021-06-18 by SuLingGG