SuLingGG/OpenWrt-Rpi #127: 使用树莓派4b当做软路由，用自带的WIFI发射无线信号，手机和电脑都可以识别并连上树莓派软路由的WIFI信号，但是另外一个树莓派4b刷的官方Raspberyy Pi OS系统开机后不能识别到这个树莓派4b软路由的无线WIFI信号，其他路由器的信号能正常识别和连网，把两个树莓派对调SD卡，那个当做Raspi OS桌面系统的树莓派还是不能识别另外一个树莓派发射的无线信号

yangwen9, 2021-04-15:
使用树莓派4b当做软路由，用自带的WIFI发射无线信号，手机和电脑都可以识别并连上树莓派软路由的WIFI信号，但是另外一个树莓派4b刷的官方Raspberyy Pi OS系统开机后不能识别到这个树莓派4b软路由的无线WIFI信号，其他路由器的信号能正常识别和连网，把两个树莓派对调SD卡，把那个当做Raspberry Pi OS桌面系统的树莓派还是不能识别另外一个树莓派发射的无线信号

EnnawYang, 2021-04-15:
改下wifi区域(国家)试试

SuLingGG, 2021-04-15:
[image]
按如图所示设置试试。
如果你点击“保存&应用”后没有出来“正在应用配置”的 Loading 框，可能需要再点击一次“保存&应用”，等待 Loading 框执行完毕后才可能完全应用配置。

yangwen9, 2021-04-15:
谢谢指点，改成2.4G的频宽后可以被另外一个树莓派4b正常识别了,但是网速下降很多，目前只有60-70M左右的速度了，应该是5G频段有冲突吧？ 按理说一个发射，一个接收，不应该啊

SuLingGG, 2021-04-15:
> 谢谢指点，改成 2.4G 的频宽后可以被另外一个树莓派 4b 正常识别了，但是网速下降很多，目前只有 60-70M 左右的速度了，应该是 5G 频段有冲突吧？ 按理说一个发射，一个接收，不应该啊

因为这样改成 2.4G 的频段后，运行 OpenWrt 的树莓派设备的 WiFi 的最大带宽只有 72Mbps。

kmzs123, 2021-04-15:
其实是树莓派的官方系统问题吧，因为官方系统把WiFi地区设置成中国之后会有一些频段被关闭，导致直接搜索不到信号，被关闭的好像是36附近的频段，你可以把官方系统的地区设置成美国的试试，或者把WiFi锁在149附近的频段
不一定是确定的这两个数，反正就是5GWiFi频段的其中一些的，你可以都试试

EnnawYang, 2021-04-15:
的确是系统问题，我之前也遇到过

yangwen9, 2021-04-15:
调整到频段149左右后，直接WIFI不能识别了，任何设备都搜索不到WiFi信号了，还有补救措施吗？

yangwen9, 2021-04-15:
如果不重写SD卡，还有什么方法能进后台呢，现在信号全没了，改了频段后

kmzs123, 2021-04-15:
你可以直接用电脑改配置文件

yangwen9, 2021-04-15:
目前对我来说有点麻烦，还是重新刷SD算了

closed 2021-04-18 by SuLingGG