SuLingGG/OpenWrt-Rpi #136: 请教，DNS Servers 以及DHCP Server怎么设定为8.8.8.8 &8.8.4.4

chan2093, 2021-04-29:
我遇到一个问题，由于IPS有过滤一些网站访问的功能，平时我都在路由器WIFI的DHCP下手动设定成 DNS server 为8.8.8.8以及8.8.4.4绕过IPS过滤器。请问OpenWrt下的DHCP修改后为什么不能更改终端接收到的DNS Servers为8.8.8.8以及8.8.4.4。
或许本人设定上有错，请求指导。谢谢。
请参阅附图
[image]
需要修改的DNS Server
[image]
手动设定但无效
[image]
需要手动设定，但手机就不好使了。

SuLingGG, 2021-04-29:
需要在
网络 - 接口 - Lan - 修改
中修改 DNS，这样 DNS 配置才能随 DHCP 直接下发到设备。

chan2093, 2021-04-29:
[image]
[image]
不好意思，我尝试了但仍然修改不到。WIFI的DHCP里的DNS仍然是192.168.1.1

SuLingGG, 2021-04-29:
重启试试。另外这些设置仅在 OpenWrt 作为 DHCP 服务器时才生效，如果你把 OpenWrt 作为不开启 DHCP 的旁路由之用，那当然不会生效。

we2oxy, 2021-04-29:
> [image]
> [image]
> 不好意思，我尝试了但仍然修改不到。WIFI的DHCP里的DNS仍然是192.168.1.1

[image]

SuLingGG, 2021-04-29:
对，也可以试试楼上 Dalao 的方法 (

chan2093, 2021-04-29:
[image]
行了，非常完美，谢谢dalao。

closed 2021-04-29 by chan2093