SuLingGG/OpenWrt-Rpi #139: A6200 信号好，但是速率上不去

wwsq1149, 2021-04-30:
有三个问题
[image]
1、如图，这个信号还是个位数的Mbps
不知道为4什么，之前是好的，可能是热插拔了一下
2、那个 通用 802.11 无线控制器 (radio0)
有的时候重新配置A6200会影响到树莓派4B自带网卡，出现未开启或不可用的提示，重新保存一两次才好，停用重开没用
3、换USB接口插的时候，会多一个radio2，原来的还在，但是变成 通用 802.11 无线控制器，设置都没了，只能固定在一个口插吗，怎么把多余的那个删掉呢

SuLingGG, 2021-04-30:
1. 我不对网卡速率做保证。

2. 我不对网卡可用性做保证。

3. 我这边能力有限，一没设备，二没能力，无法对千奇百怪的网卡设备做支持和优化。

4. 热插拔确实会造成这个问题，如果你想删除空闲的“控制器”，需要手动编辑 /etc/config/wireless 文件，删除对应 radio* 的配置代码块，保存文件即可将空闲的控制器删除。

5. 我还是建议使用我文档里推荐的无线网卡。

wwsq1149, 2021-04-30:
@SuLingGG sorry打错了，是你推荐的A6210

SuLingGG, 2021-04-30:
检查一下你是否用了 USB 延长线，如果用了的话 USB 口直插网卡试试。
另外不建议在 OpenWrt 里对网卡进行 802.x 之类的高级配置，可能会造成各种各样奇怪的问题。
无线加密方式设为 wpa2 即可，或者不加密试试。

EnnawYang, 2021-04-30:
可以试试删除 `/etc/config/wireless` 并重启能不能恢复正常，内置与外置无线尽量只使用其中一个

wwsq1149, 2021-04-30:
@EnnawYang 试过没用
现在重新刷今天的最新固件
md内置无线改成Client连个网再改发现改不回热点了。。。
重启什么的，删wireless都用过了
已经炸毛了。哪个版本或者哪一天的固件比较稳定的

wwsq1149, 2021-04-30:
@SuLingGG 哪一天版本对于A6210的支持比较好并且比较稳定的？

SuLingGG, 2021-04-30:
> @SuLingGG 哪一天版本对于 A6210 的支持比较好并且比较稳定的？

这个问题你问我我也不知道...

wwsq1149, 2021-04-30:
@SuLingGG emm好吧，还是感谢

closed 2021-04-30 by wwsq1149