SuLingGG/OpenWrt-Rpi #140: 能否在Rpi把luci-app-statistics默认加上，谢谢大佬

galaxyskyknight, 2021-05-03:
这个插件对我很有用，但是Rpi固件没有包含，自己编译又麻烦，一般也不想去折腾Mini.  而且自己编出来的ipk安装到Rpi固件上有问题，graphic显示不出来，运行日志提示缺少 libiwinfo20210106.so.的依赖库。谢谢

Nlqqer, 2021-05-03:
@galaxyskyknight 一般没人用的插件不会单独加上，如果有需要可以自行编译

galaxyskyknight, 2021-05-04:
说了自行编译的上传后系统提示缺少libiwinfo20210106.so, 应该是跟Rpi的固件有依赖缺失。

Nlqqer, 2021-05-04:
@galaxyskyknight 我说的是自行编译固件...

galaxyskyknight, 2021-05-04:
> @galaxyskyknight 我说的是自行编译固件...

自行编译有部分插件没法做到跟rpi的完全对齐。。。。长期使用rpi，习惯了，有些ipk在coolsnowwolf的luci库里没有，也不知道哪里找

lxl6125, 2021-05-04:
这东西仓库里不是有吗？

galaxyskyknight, 2021-05-04:
> 这东西仓库里不是有吗？

好累....1)仓库是有，但Rpi固件没编译上，没勾选。2）如果我自己整体编译固件，比对Rpi固件，Rpi固件里有些其他我常用的插件找不到源没法编出来跟Rpi的一样。3）我自己如果单独编译statistics的ipk插件区安装，编译出来安装到Rpi固件上，提示缺少libiwinfo20210106.so,  这样说明白了吗？

lxl6125, 2021-05-04:
> > 这东西仓库里不是有吗？
> 
> 好累....1)仓库是有，但Rpi固件没编译上，没勾选。2）如果我自己整体编译固件，比对Rpi固件，Rpi固件里有些其他我常用的插件找不到源没法编出来跟Rpi的一样。3）我自己如果单独编译statistics的ipk插件，编译出来安装到Rpi固件的live系统上，提示缺少libiwinfo20210106.so, 这样说明白了吗？

不明白，既然仓库有，安装一下就行了啊

closed 2021-06-18 by SuLingGG