SuLingGG/OpenWrt-Rpi #142: root分区扩容

kexix, 2021-05-04:
您好，看了您的文档之后，按照您的操作方法，只能对 rootfs分区扩容，不能对root分区扩容，想对root分区也进行一下扩容，因为SD卡是128G的，不想浪费了

Nlqqer, 2021-05-04:
@kexix root分区要装什么东西？

kexix, 2021-05-04:
> @kexix root分区要装什么东西？

因为太小总感觉不够用，比如升级固件的时候，对boot分区要升级的话就觉得不行，还有就是有时候安装自己的一些插件会习惯的上次到root下

SuLingGG, 2021-05-04:
@kexix rootfs 分区不是 root 分区吗...

kexix, 2021-05-04:
> @kexix rootfs 分区不是 root 分区吗...

不晓得，这个是怎么区分，因为在Disk之类的分区软件里看见有，root分区

zzcabc, 2021-05-04:
如果你想改，请自行fork，改配置

kexix, 2021-05-04:
> 如果你想改，请自行fork，改配置

好的，谢谢，我正在尝试

galaxyskyknight, 2021-05-06:
> 如果你想改，请自行fork，改配置

配置在哪个文件里.....?

kexix, 2021-05-06:
> > 如果你想改，请自行fork，改配置
> 
> 配置在哪个文件里.....?

树莓派

kexix, 2021-05-06:
> > > 如果你想改，请自行fork，改配置
> > 
> > 
> > 配置在哪个文件里.....?
> 
> 树莓派

config，树莓派额外配置里，经过我的尝试，可以生成10G的镜像，再大会提示编译机空间，会失败，而且建议只生成ext4格式就够了，因为超过6小时也会编译失败

closed 2021-05-06 by kexix

galaxyskyknight, 2021-05-06:
> > > > 如果你想改，请自行fork，改配置
> > > 
> > > 
> > > 配置在哪个文件里.....?
> > 
> > 
> > 树莓派
> 
> config，树莓派额外配置里，经过我的尝试，可以生成10G的镜像，再大会提示编译机空间，会失败，而且建议只生成ext4格式就够了，因为超过6小时也会编译失败

make config哪个位置指定？