SuLingGG/OpenWrt-Rpi #144: shadowsocks无法转发udp数据

le520, 2021-05-12:
在使用shadowsocks-plus的时候，udp数据无法被正常的转发。
经过测试已经确认我的节点支持udp，并且在我的路由器上挂上ssr也是可以正常玩游戏的。这似乎是一个比较常见的问题，但没有找到解决方案。

le520, 2021-05-12:
补充一下，已经开启游戏模式。

Nlqqer, 2021-05-12:
@le520 去找插件开发者反馈，这里只解决固件问题

le520, 2021-05-12:
@Nlqqer 这不是插件的问题，在路由器上和部分RPI的ssr上是可以正常转发udp的数据的，因此我怀疑是缺少了一些东西。（我自己也手动编译过但udp转发也不生效）

le520, 2021-05-12:
@Nlqqer 我的路由器也是用的openwrt，挂上ssr打游戏是可以正常走代理的，但是rpi上的就不能用，我测试了恩山论坛上的几乎所有固件，只找到了一款可以正常用ssr转发udp的。

EnnawYang, 2021-05-12:
这确实是插件问题

Nlqqer, 2021-05-12:
@le520 你用的什么插件？

le520, 2021-05-12:
@Nlqqer shadowsocks-plus和passwall都试过，都没办法正常使用。（我的节点的协议是shadowsocks）

le520, 2021-05-12:
> 这确实是插件问题

目前有什么解决方案吗？

le520, 2021-05-12:
@Nlqqer @EnnawYang 
以下是我目前在用的固件中的与shadowsocks相关的软件包和版本号（测试可以正常转发udp数据）。
 shadowsocks-libev-ss-local | 3.3.4-1
 shadowsocks-libev-ss-redir | 3.3.4-1
 shadowsocksr-libev-alt | 2.5.6-5
 shadowsocksr-libev-server | 2.5.6-5
 shadowsocksr-libev-ssr-local | 2.5.6-5

Nlqqer, 2021-05-12:
@le520 你有没有装docker

le520, 2021-05-12:
@Nlqqer 可以转发udp的固件里是没有装docker的，比较精简。

Nlqqer, 2021-05-12:
@le520 docker会影响udp转发，如果必须要docker的话用openclash的tun模式

le520, 2021-05-12:
> @le520 docker会影响udp转发，如果必须要docker的话用openclash的tun模式

好像确实是docker的问题，我使用纯净版手动装了ssr，测试了一下udp是可以正常转发的。

closed 2021-05-12 by le520