SuLingGG/OpenWrt-Rpi #146: zerotier引起系统负载过高

Licsber, 2021-05-14:
系统负载变化曲线如下：
[image]

出问题的现象：以下进程大量增加
sh -c config .... /etc/zerotier.start
目前没找到在哪里出现的这个调用
截图如下
[image]

解决方案 killall sh
强行终止所有sh进程
目前暂时无法复现 希望有路过的大佬看看哪里出了问题

SuLingGG, 2021-05-14:
建议你去 https://github.com/coolsnowwolf/lede/issues 反馈一下。

Licsber, 2021-05-15:
[image]
更新：
冤枉好人了 问题出在zerotier上
/etc/zerotier.start脚本有问题
会卡在iptables的配置上
由于mwan3的ifup事件会调用fw3 -q reload
其会调用/etc/zerotier.start 导致所有hotplug都卡在这里
目前解决方案为关闭zerotier
此issue等待上游解决

closed 2021-06-18 by SuLingGG