SuLingGG/OpenWrt-Rpi #147: 请教3B+ WiFi 频繁中断

nemo2018, 2021-05-17:
各位好，
今天使用最新的固件，树莓派3B+上刷好，计划做旁路由，但是要用树莓派自带的WiFi做热点。
只打开openclash，其他都不动。
Su大的固件已经使用在其他设备使用很久，openclash使用很熟悉，配置好启动都正常。

使用了一阵发现每隔几分钟，电脑的WiFi就要断一下，去看了日志发现有如下不正常记录。
试着Google了一下，尝试了几个地方都没有改善。
请问我们有同样使用场景的朋友遇到类似问题么？
谢谢！
[image]

SuLingGG, 2021-05-18:
你有可能启用了无线加密并且启用了一些高级的 802.11x 特性。
试试关闭无线加密，或者尝试将无线加密相关设置修改为如图所示设置:

[image]

在某些设备上启用无线加密会出现一些不可预知的问题，建议不为热点设置密码。如果必要，可以将 SSID 隐藏。

nemo2018, 2021-05-18:
感谢大佬回答。我按照上述建议的设置了无线加密，依旧。
不过没关系了，我改变了用法，用无线做了wwan，lan接电脑上，反正我只是给一台有网口的电脑用而已。
再次感谢！

closed 2021-06-18 by SuLingGG