SuLingGG/OpenWrt-Rpi #152: 固件干净不 ？

capewang, 2021-06-05:
固件干不干净啊，   不使用软路由时候，     家庭原ip会不会被当成机器人？   出验证码？

lxl6125, 2021-06-05:
@capewang 说不干净，你信吗？
说干净，你又信吗？

Nlqqer, 2021-06-05:
@capewang 这个问题，不用就可以避免，或者你写一个openwrt替代品也不是不行

capewang, 2021-06-05:
> @capewang 说不干净，你信吗？
> 说干净，你又信吗？

我信不信你酸个毛线哦。

capewang, 2021-06-05:
> @capewang 这个问题，不用就可以避免，或者你写一个openwrt替代品也不是不行
issue 不能问啊，还是搓到谁心了。

capewang, 2021-06-05:
是这样，前几天在海外服务器装了一个。 我都没在国内用过。  结果被国内给墙了。

难道就因为22端口，就给我墙了啊？  不太可能是吧。
那定然是有国内连接到我服务器了，连接还不会少的，才封掉。 
国内又不用，还被国内封了。   
估计被人拿去爬虫，seo优化去了。  
所以我得问一下喽。

capewang, 2021-06-05:
想过设置防火墙 ，但是没用

限制ip进站， 肯定是很多动态ip进站，卖出去得爬虫网络，不会是一个人使用。 

并且即便将国内ip段全填 进去，也没用，还是会经过国内防火墙，到主机再拒绝。。。。。

capewang, 2021-06-05:
装原版的，ssr plus 资源 搞得好烦哦，没一个正常的的。

closed 2021-06-05 by capewang

Nlqqer, 2021-06-06:
@capewang 知道前几天什么日子吗，国内中转机都被墙了

SuLingGG, 2021-06-06:
> 说不干净，你信吗？
> 说干净，你又信吗？

@lxl6125 这位说得没错。我是不明白为啥我已经开源了还会有人问这个...

固件所用源码绝大多数采用开源项目编译，即使包含部分未开源的第三方包，那也是经过众多用户的验证并经过慎重考虑才添加的。

固件基于 Github Action 编译，源码开源，编译流程文件开源，所用脚本开源，机器由 Github Action 提供。Github Action 将根据上述文件每日自动编译，你可以自行审阅每一行代码。

本项目和部分提供成品文件并且开源的项目还不一样，成品文件和开源项目所用源码谁也不能保证是绝对一致的，也就是说，成品文件存在夹带私货的可能，但基于 Github Action 编译的成品文件是纯粹的所见即所得的。只要源码/流程文件/脚本是唯一的，那么编译出来的固件可以说就是唯一的，不存在夹带私货的可能。

综上，清者自清吧。

luoxinyu2002, 2021-11-14:
> 装原版的，ssr plus 资源 搞得好烦哦，没一个正常的的。

你爱用不用