SuLingGG/OpenWrt-Rpi #162: 新版固件不识别微软1663网卡

ahfncj, 2021-06-25:
一直在用3-12的固件   可以识别微软1663网卡,今天升级到最新固件,无法识别微软网卡了

SuLingGG, 2021-08-18:
opkg update
opkg remove kmod-usb-net-rtl8152-vendor
opkg install kmod-usb-net-rtl8152
试试。

ahfncj, 2021-08-18:
> opkg
> updatekg remove kmod-usb-net-rtl8152-vendor
> opkg install kmod-usb-net-rtl8152

可以了 老哥    下版固件会修复吗

moshouhot, 2021-09-01:
树莓派插的8153B，过几个小时就断网重连，最近几个月都有这个问题，今天终于破案了。换了网卡驱动，立马就好了。还以为8153B网卡坏了，差点就去买88179网卡。恳请大神早日修复这个BUG。

xqdxqd, 2021-11-01:
我的usb网卡也会不定期断开，重启后恢复

Mon Nov  1 02:14:52 2021 kern.info kernel: [35164.561961] NETDEV WATCHDOG: eth1 (ax88179_178a): transmit queue 0 timed out
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.569092] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:473 dev_watchdog+0x2b8/0x2c0
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.577358] Modules linked in: xt_FULLCONENAT ath9k_htc ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib rt2500usb pppoe ppp_async option mt76x0u mt76x0_common l2tp_ppp cdc_mbim carl9170 brcmfmac ath9k_hw ath6kl_usb ath6kl_core ath zd1211rw wireguard uvcvideo usb8xxx usb_wwan sr9700 snd_usb_audio smsc95xx sierra_net rtlwifi rtl88x2bu rtl8821cu rtl8812au rtl8188eu rtl8187 rt73usb rt2x00usb rt2x00lib rsi_usb rsi_91x rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic plusb p54usb p54common mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615_common mt7603e mt7601u mt76_usb mt76_connac_lib mt76 mcs7830 mac80211 libertas libchacha20poly1305 libblake2s kalmia ipt_REJECT huawei_cdc_ncm dm9601 chacha_neon cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a asix zstd xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.577494]  xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_cgroup xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY videobuf2_v4l2 videobuf2_common usbserial usbnet usblp usbhid ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr snd_usbmidi_lib slhc sch_cake rtl8150 r8712u(C) r8152 poly1305_neon pegasus nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo libcurve25519_generic libchacha libblake2s_generic lib80211_crypt_wep
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.665288]  lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables hwmon hso hid_generic crc_itu_t crc_ccitt compat cdc_wdm cdc_acm brcmutil asn1_decoder fuse sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact snd_bcm2835(C) videobuf2_vmalloc videobuf2_memops hid videodev evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink net2280 ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.752561]  ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfsd nfs bonding ip_gre gre ifb dummy sit l2tp_netlink l2tp_core ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel rpcsec_gss_krb5 auth_rpcgss oid_registry veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm_dmaengine snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore xfrm_user xfrm_ipcomp af_key xfrm_algo lockd sunrpc grace cifs dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 md_mod nls_utf8 eeprom_93cx6 vxlan udp_tunnel ip6_udp_tunnel crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic md5 md4 kpp ghash_generic gf128mul gcm echainiv des_generic libdes deflate cmac ccm authenc arc4 crypto_acompress vfat fat nls_iso8859_1 nls_cp437 ohci_pci uhci_hcd ohci_platform ohci_hcd ehci_pci fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd exfat(C) btrfs xor
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.840390]  zstd_decompress zstd_compress zlib_inflate zlib_deflate xxhash xor_neon raid6_pq lzo_decompress lzo_compress libcrc32c antfs encrypted_keys trusted tpm
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.942618] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.155 #0
Mon Nov  1 02:14:52 2021 kern.warn kernel: [35164.950014] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)