SuLingGG/OpenWrt-Rpi #170: R2S反馈

463978209, 2021-07-08:
OLED功能无法正常使用

SuLingGG, 2021-07-09:
我这边暂时没办法解决...
或者说你那边用谁的固件 OLED 可以正常显示那可以放一下他的项目地址我参考下~
(因为我这边没有 R2S)

463978209, 2021-07-09:
https://github.com/soffchen/NanoPi-R2S
这位仁兄的还行，就是固件温度太高了   WiFi驱动不太全（我用的太偏门了）

SuLingGG, 2021-07-09:
但是我的和他的项目用的 OLED 源码是同一个...这就有点无解...

463978209, 2021-07-09:
源码项目里面有个解答说是内核驱动不完整.....

SuLingGG, 2021-07-09:
我也注意到了，在 https://github.com/SuLingGG/OpenWrt-Rpi/commit/970516c93195973d991ec81363dc18080e6ebaff 添加了额外的 i2c 驱动，下版固件编译成功的话可以试试看~
或者你可以在当前固件下尝试运行以下命令:

```
opkg update && opkg install  kmod-i2c-algo-bit kmod-i2c-algo-pca kmod-i2c-algo-pcf kmod-i2c-core kmod-i2c-designware-pci kmod-i2c-gpio kmod-i2c-mux kmod-i2c-mux-gpio kmod-i2c-mux-pca9541 kmod-i2c-mux-pca954x kmod-i2c-pxa kmod-i2c-smbus kmod-i2c-tiny-usb
```
然后重启试试。

463978209, 2021-07-09:
当前固件TTYD运行保存重启后问题依旧，OLED提示未运行。期待下一版本
非常感谢 “隔壁家的叔叔“  哈哈

closed 2021-11-16 by AmadeusGhost