SuLingGG/OpenWrt-Rpi #205: 树莓派4B随机重启如何排查？

do21, 2021-08-28:
大佬你好！
我有一台树莓派4B 4G v1.2，用Raspi OS的话，可以正常运行好几天，但是刷了大佬的固件，除了网口的IP没有任何别的设置，大概十多个小时会自动重启一次，但是系统里面也没有Crashlog，我看了一下Cron也是默认的三个任务，目前不知道怎么排查，恳请大佬指教，谢谢！！

SuLingGG, 2021-08-28:
一般来说是供电的问题。最好不要用快充头给树莓派供电。如果 USB 接着外置设备，比如硬盘盒，无线网卡等也会造成供电不足...

do21, 2021-08-28:
> 一般来说是供电的问题。最好不要用快充头给树莓派供电。如果 USB 接着外置设备，比如硬盘盒，无线网卡等也会造成供电不足...

我用的是Nexus 5X的LG 5V3A Type-C充电头，没有快充也没有其他电压挡位，我也试过在1688上买的一个只有5V3A一档的充电头，发现还是会有这个情况。
外置USB部分我用了一个USB HUB，外接5V2A电源，只插了键盘和鼠标的接收器。

测试时候，HDMI接了显示器，USB接了外置供电的USB HUB插着键盘鼠标接收器，RJ45接着网线，就没有其它设备了，因为Raspi OS能持续运行，可能会有其他方面的问题吗？不知有没有办法打开Crashlog以便提交issue分析呢？

谢谢！

SuLingGG, 2021-08-28:
没有什么好的办法...
这个和系统关系不大，因为 OpenWrt 和官方系统很多地方都有差异。
但还是建议你把不必要的外接设备 (HDMI/USB) 拔掉，再测试一段时间。

closed 2021-09-16 by SuLingGG