SuLingGG/OpenWrt-Rpi #206: 大佬可否添加一下kmod-can

misakiesl, 2021-08-29:


SuLingGG, 2021-08-29:
opkg update && opkg install kmod-can

misakiesl, 2021-08-29:
[image]
[image]

我这会出现Unknown package 'kmod-can'.

misakiesl, 2021-08-29:
> opkg update && opkg install kmod-can

[image]
[image]
我按您说的这样做和下载下来进行安装都会出现Unknown package 'kmod-can'.这个该怎么办啊

SuLingGG, 2021-09-07:
现在应该解决了这个问题。下载最新版固件试一下，按如上方法安装 kmod 包就行。

misakiesl, 2021-09-15:
> 现在应该解决了这个问题。下载最新版固件试一下，按如上方法安装 kmod 包就行。

成功了，谢谢大佬

closed 2021-09-16 by SuLingGG