SuLingGG/OpenWrt-Rpi #207: at out and smstool

iptv-almanet, 2021-09-04:
Hi,i just install ipk for github and have an eror
can you compile for  rasbery pi 4 openwrt?
here the source :
at out : https://github.com/4IceG/luci-app-atinout-mod
smstool : https://github.com/4IceG/luci-app-sms-tool
thanks for advice
[image]

zzcabc, 2021-09-04:
Opkg update before installation, but at out may not，or using [openwrt-mini](https://github.com/SuLingGG/OpenWrt-Mini)

SuLingGG, 2021-09-05:
All the LuCI APPS are "user interface" packages, so user interface packages are not included binary core.
If you try to install UI packages without install binary packages, opkg will throw a "architectures error", it is the bug of opkg, this error is always caused by "dependences missing"
luci-app-atinout-mod depends on package atinout，luci-app-sms-tool depends on package sms-tool. You shoud install atinout-xxx.ipk or sms-tool-xxx.ipk before luci-app-xxx.ipk, or put luci-app & binary packages into a folder then install them by "opkg install *.ipk".
So you should find the right ipk packages for your device, I noticed that the owner of repository did not provide ipk packages for sms-tool or atinout, maybe you have to build these binary packages by yourself.

closed 2021-09-07 by SuLingGG