SuLingGG/OpenWrt-Rpi #209: About sysupdate

18rudi, 2021-09-08:
I trying to update fw from menu>>backup/flash firmware but not working
Can you add package tiny file manager?
Thanks 🙏

SuLingGG, 2021-09-08:
The firmware of OpenWrt-Rpi can not upgrade due to some compatibility issues.
Maybe you can try the firmware of [OpenWrt-Mini](https://github.com/SuLingGG/OpenWrt-Mini), firmwares of this repository can upgrade at LuCI successfully.

18rudi, 2021-09-09:
OpenWrt-Mini missing proto qmi n modemmanager

arminmacx, 2021-09-10:
so how can we upgrade to latest version of compiled from github? i had same issue each time I try to use sysupgrade it just went through but then without any reboot it backed to main interface and even after manually reboot it just shows same build and date for the firmware

SuLingGG, 2021-09-10:
> so how can we upgrade to latest version of compiled from github? i had same issue each time I try to use sysupgrade it just went through but then without any reboot it backed to main interface and even after manually reboot it just shows same build and date for the firmware

Sorry, not support now...🥺🥺

SuLingGG, 2021-09-10:
> OpenWrt-Mini missing proto qmi n modemmanager

You can install them by opkg.
The firmware of OpenWrt-Mini can install packages what you like by opkg.

closed 2021-09-16 by SuLingGG