SuLingGG/OpenWrt-Rpi #232: 最近取消了wetransfer后，fork过去自行修改了配置，就没办法上传下载新生成的固件了

ricksuzade-maker, 2021-09-22:
如果作者愿意的话，可否开个教程让大家可以把自定义后的固件传到自己的空间？

zzcabc, 2021-09-22:
https://p3terx.com/archives/build-openwrt-with-github-actions.html 看自定义环境变量，能否帮助你

SuLingGG, 2021-09-22:
@ricksuzade-maker 已在 90e236e3de52ca5d03b6d5d91de9eee8ffc68841 加入了上传固件至 Release 的步骤，不过此步骤默认未启用，如果你 Fork 了本项目，可以在 Github 内 Fetch upstream 后，在 `.github/workflows` 目录下把指定设备的 yml 文件中的:

`UPLOAD_RELEASE: false`

改为

`UPLOAD_RELEASE: true`

然后再触发编译即可。编译完成后会自动将编译生成物上传至 Release。

ricksuzade-maker, 2021-09-22:
上午跑了一个进程，下午看结果似乎release空间太小，放不下最终生成的固件

ricksuzade-maker, 2021-09-22:
<div class="mt-1 mb-3 px-3" style="box-sizing: border-box; margin-top: 4px !important; margin-bottom: 16px !important; padding-right: 16px !important; padding-left: 16px !important; color: rgb(201, 209, 217); font-family: -apple-system, system-ui, &quot;Segoe UI&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 14px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="mx-0 mx-md-1" style="box-sizing: border-box; margin-right: 4px !important; margin-left: 4px !important;"><h4 class="text-bold" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 0px; font-size: 16px; font-weight: 600 !important;">Annotations</h4><div class="color-text-secondary text-small" style="box-sizing: border-box; font-size: 12px !important; color: var(--color-fg-muted)  !important;">1 error and 1 warning</div></div></div>

BuildSystem.IO.IOException: No space left on device    at System.IO.FileStream.WriteNative(ReadOnlySpan`1 source)    at System.IO.FileStream.FlushWriteBuffer()    at System.IO.FileStream.Flush(Boolean flushToDisk)    at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)    at System.Diagnostics.TextWriterTraceListener.Flush()    at GitHub.Runner.Common.HostTraceListener.WriteHeader(String source, TraceEventType eventType, Int32 id)    at GitHub.Runner.Common.HostTraceListener.TraceEvent(TraceEventCache eventCache, String source, TraceEventType eventType, Int32 id, String message)    at System.Diagnostics.TraceSource.TraceEvent(TraceEventType eventType, Int32 id, String message)    at GitHub.Runner.Worker.Worker.RunAsync(String pipeIn, String pipeOut)    at GitHub.Runner.Worker.Program.MainAsync(IHostContext context, String[] args) System.IO.IOException: No space left on device    at System.IO.FileStream.WriteNative(ReadOnlySpan`1 source)    at System.IO.FileStream.FlushWriteBuffer()    at System.IO.FileStream.Flush(Boolean flushToDisk)    at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)    at System.Diagnostics.TextWriterTraceListener.Flush()    at GitHub.Runner.Common.HostTraceListener.WriteHeader(String source, TraceEventType eventType, Int32 id)    at GitHub.Runner.Common.HostTraceListener.TraceEvent(TraceEventCache eventCache, String source, TraceEventType eventType, Int32 id, String message)    at System.Diagnostics.TraceSource.TraceEvent(TraceEventType eventType, Int32 id, String message)    at GitHub.Runner.Common.Tracing.Error(Exception exception)    at GitHub.Runner.Worker.Program.MainAsync(IHostContext context, String[] args) Unhandled exception. System.IO.IOException: No space left on device    at System.IO.FileStream.WriteNative(ReadOnlySpan`1 source)    at System.IO.FileStream.FlushWriteBuffer()    at System.IO.FileStream.Flush(Boolean flushToDisk)    at System.IO.StreamWriter.Flush(Boolean flushStream, Boolean flushEncoder)    at System.Diagnostics.TextWriterTraceListener.Flush()    at System.Diagnostics.TraceSource.Flush()    at GitHub.Runner.Common.TraceManager.Dispose(Boolean disposing)    at GitHub.Runner.Common.TraceManager.Dispose()    at GitHub.Runner.Common.HostContext.Dispose(Boolean disposing)    at GitHub.Runner.Common.HostContext.Dispose()    at GitHub.Runner.Worker.Program.Main(String[] args)Show less
--
BuildYou are running out of disk space. The runner will stop working when the machine runs out of disk space. Free space left: 0 MB

ricksuzade-maker, 2021-09-30:
最近suling官方的action里，只有google drive的链接了，但是进去看只更新到9月10日，后面的虽然编译成功了，但是却看不到存到哪里去了。

closed 2021-11-16 by AmadeusGhost