SuLingGG/OpenWrt-Rpi #235: AdguardHome

Fregionz, 2021-09-30:
Anyone have problem with enabling AdguardHome? it said AdGuardHome RUNNING and Redirected already but my network can't detected in Adguard Log and the DNS not redirected into Adguard. Thank you
[image]

Feeling1016, 2021-09-30:
You have to change your Dns too in Adg. Yours is 5553. Change it to 53.

Fregionz, 2021-09-30:
The issue solved after i changed on Firewall - Custom Rules, by deleting custom port 53 redirect, i will close this issue, thank yo
[image]
u

closed 2021-09-30 by Fregionz