SuLingGG/OpenWrt-Rpi #243: 安装tcpdump提示  Read-only file system.

leechaowen, 2021-11-03:
用系统帐号root 安装tcpdump提示没有权限，如何解决？谢谢抽空解答。

$ opkg install tcpdump
Installing tcpdump (4.9.3-3) to root...
Downloading https://mirrors.cloud.tencent.com/openwrt/snapshots/packages/aarch64_cortex-a72/base/tcpdump_4.9.3-3_aarch64_cortex-a72.ipk
Collected errors:
 * wfopen: //usr/lib/opkg/info/tcpdump.control: Read-only file system.
 * wfopen: //usr/lib/opkg/info/tcpdump.postinst: Read-only file system.
 * wfopen: //usr/lib/opkg/info/tcpdump.prerm: Read-only file system.
 * wfopen: /usr/sbin/tcpdump: Read-only file system.
 * pkg_write_filelist: Failed to open //usr/lib/opkg/info/tcpdump.list: Read-only file system.
 * opkg_install_pkg: Failed to extract data files for tcpdump. Package debris may remain!
 * opkg_install_cmd: Cannot install package tcpdump.

leechaowen, 2021-11-03:
[image]

AmadeusGhost, 2021-11-03:
你的系统只读了，重刷吧

leechaowen, 2021-11-03:
[image]

可以通过磁盘管理 修改只读属性吗？

leechaowen, 2021-11-03:
$ df -Th
Filesystem           Type            Size      Used Available Use% Mounted on
/dev/root            ext4          944.9M    908.7M     20.2M  98% /
tmpfs                tmpfs           1.9G     16.4M      1.8G   1% /tmp
/dev/mmcblk0p1       vfat           63.9M     16.4M     47.4M  26% /boot
tmpfs                tmpfs         512.0K         0    512.0K   0% /dev
/dev/mmcblk0p3       ext2           28.3G    925.2M     25.9G   3% /mnt/mmcblk0p3
/dev/mmcblk0p3       ext2           28.3G    925.2M     25.9G   3% /mnt/mmcblk0p3/docker

leechaowen, 2021-11-03:
[image]

leechaowen, 2021-11-03:
[image]

ligusx, 2021-11-03:
终端运行以下命令后重启即可，如需彻底解决，请重刷
e2fsck -y /dev/mmcblk0p2

closed 2021-11-16 by AmadeusGhost