SuLingGG/OpenWrt-Rpi #247: 树莓派3B，我无论怎么点保存或者应用都不能更改我的配置

nctqaq, 2021-11-04:
[image]
[image]
[image]
例如SSID，WiFi加密方式和WiFi密码，更改后点击“保存&应用”都会变回OpenWrt和无密码
例如lan接口取消绑定eth0，添加wan口，都不能实现，只会跳回新刷机最初始默认的配置
我只有路由器管理密码更改成功了...

nctqaq, 2021-11-04:
[image]
[image]

nctqaq, 2021-11-04:
右上角“未保存的配置”，点保存或者保存&应用，没反应，比如13条未保存，之后还是13条未保存

AmadeusGhost, 2021-11-05:
显然你的系统只读了，重刷吧。

closed 2021-11-16 by AmadeusGhost