SuLingGG/OpenWrt-Rpi #257: 固件内部 `/sbin/cpuinfo` 语法出现问题、导致系统日志一只刷报错

expoli, 2021-11-16:
- 系统版本
OpenWrt R21.11.11 (2021-11-13) / LuCI Master (git-21.280.14233-9b796b1)
- 报错信息
```log
Tue Nov 16 09:14:04 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:14:09 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:14:14 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:14:19 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:14:24 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:14:29 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:15:36 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
Tue Nov 16 09:16:00 2021 daemon.err uhttpd[4060]: /sbin/cpuinfo: line 25: syntax error: unexpected "else"
```

- 脚本信息
- 语法出现问题、缺少对应的语法对
```bash
  1 #!/bin/sh
  2 
  3 cpu_arch="$(cat "/proc/cpuinfo" | grep "model name" | sed -n "1p" | awk -F ': ' '{print $2}')"
  4 [ -z "${cpu_arch}" ] && cpu_arch="ARMv8 Processor"
  5 cpu_cores="$(cat "/proc/cpuinfo" | grep "processor" | wc -l)"
  6 
  7 if grep -q "bcm27xx" "/etc/openwrt_release"; then
  8         cpu_freq="$(expr $(vcgencmd measure_clock arm | awk -F '=' '{print $2}') / 1000000)Mhz"
  9 elif grep -q "bcm53xx" "/etc/openwrt_release"; then
 10         cpu_freq="$(nvram get clkfreq | awk -F ',' '{print $1}')MHz"
 11 elif grep -q "mvebu" "/etc/openwrt_release"; then
 12         cpu_freq="$(cat "/proc/cpuinfo" | grep "BogoMIPS" | sed -n "1p" | awk -F ': ' '{print $2}')MHz"
 13 else
 14         cpu_freq="$(expr $(cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq) / 1000)MHz"
 15         big_cpu_freq="$(expr $(cat /sys/devices/system/cpu/cpufreq/policy4/cpuinfo_cur_freq 2>"/dev/null") / 1000 2>"/dev/null")"
 16         [ -n "${big_cpu_freq}" ] && big_cpu_freq="${big_cpu_freq}MHz "
 17 fi
 18 
 19 if grep -q "ipq40xx" "/etc/openwrt_release"; then
 20         sys_temp="$(awk '{ printf("%.1f°C ", $0 / 1000) }' /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input)"
 21 else
 22         sys_temp="$(awk '{ printf("%.1f°C ", $0 / 1000) }' /sys/class/thermal/thermal_zone0/temp)"
 23 fi
 24         echo -n "${cpu_arch} x ${cpu_cores} (${cpu_freq}, ${sys_temp})"
 25 else
 26         if grep -q "bcm27xx" "/etc/openwrt_release"; then
 27                 cpu_temp="$(vcgencmd measure_temp | awk -F '=' '{print $2}' | awk -F "'" '{print $1}')°C"
 28         else
 29                 cpu_temp="$(awk "BEGIN{printf (\"%.1f\n\",$(cat /sys/class/thermal/thermal_zone0/temp)/1000) }")°C"
 30         fi
 31         echo -n "${cpu_arch} x ${cpu_cores} (${big_cpu_freq}${cpu_freq}, ${cpu_temp})"
 32 fi
```

AmadeusGhost, 2021-11-16:
那个骷髅头干的好事，你更新最新的固件就可以了。

closed 2021-11-16 by AmadeusGhost

expoli, 2021-11-16:
~~我想着我已经改好了~~ 你行动力也太快了

```bash
#!/bin/sh

cpu_arch="$(cat "/proc/cpuinfo" | grep "model name" | sed -n "1p" | awk -F ': ' '{print $2}')"
[ -z "${cpu_arch}" ] && cpu_arch="ARMv8 Processor"
cpu_cores="$(cat "/proc/cpuinfo" | grep "processor" | wc -l)"

if grep -q "bcm27xx" "/etc/openwrt_release"; then
        cpu_freq="$(expr $(vcgencmd measure_clock arm | awk -F '=' '{print $2}') / 1000000)Mhz"
elif grep -q "bcm53xx" "/etc/openwrt_release"; then
        cpu_freq="$(nvram get clkfreq | awk -F ',' '{print $1}')MHz"
elif grep -q "mvebu" "/etc/openwrt_release"; then
        cpu_freq="$(cat "/proc/cpuinfo" | grep "BogoMIPS" | sed -n "1p" | awk -F ': ' '{print $2}')MHz"
else
        cpu_freq="$(expr $(cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq) / 1000)MHz"
        big_cpu_freq="$(expr $(cat /sys/devices/system/cpu/cpufreq/policy4/cpuinfo_cur_freq 2>"/dev/null") / 1000 2>"/dev/null")"
        [ -n "${big_cpu_freq}" ] && big_cpu_freq="${big_cpu_freq}MHz "
fi

if grep -q "ipq40xx" "/etc/openwrt_release"; then
        sys_temp="$(awk '{ printf("%.1f°C ", $0 / 1000) }' /sys/class/ieee80211/phy*/device/hwmon/hwmon*/temp1_input)"
else
        sys_temp="$(awk '{ printf("%.1f°C ", $0 / 1000) }' /sys/class/thermal/thermal_zone0/temp)"
fi

if grep -q "bcm27xx" "/etc/openwrt_release"; then
        cpu_temp="$(vcgencmd measure_temp | awk -F '=' '{print $2}' | awk -F "'" '{print $1}')°C"
else
        cpu_temp="$(awk "BEGIN{printf (\"%.1f\n\",$(cat /sys/class/thermal/thermal_zone0/temp)/1000) }")°C"
fi

if ${big_cpu_freq}; then
        echo -n "${cpu_arch} x ${cpu_cores} (${big_cpu_freq}${cpu_freq}, ${cpu_temp})"
else
        echo -n "${cpu_arch} x ${cpu_cores} (${cpu_freq}, ${sys_temp})"
fi
```