SuLingGG/OpenWrt-Rpi #259: 建议添加对不同文件系统的支持

Youkii-Chen, 2021-11-17:
您的固件没有添加对一些常见文件系统的支持, 在连接硬盘的时候会遇到开机无法启动的问题.

SuLingGG, 2021-11-17:
为什么笃定是我固件的问题而不从供电上找找原因呢...

Youkii-Chen, 2021-11-17:
因为检查了下你的固件里并没有添加这些
我用的是x64的固件，不存在供电问题

Youkii-Chen, 2021-11-17:
插入硬盘后，会在开机时受屏明确提示无法识别某硬盘的文件操作系统，导致进不去系统。
使用opkg安装的模块会在启动后加载进内核，但是识别硬盘这个操作在内核刚启动就已经执行了，所以猜测必须得编译时勾选启用才有效。

SuLingGG, 2021-11-17:
所以你的硬盘是啥文件系统...

SuLingGG, 2021-11-17:
opkg list-installed | grep kmod-fs
可以列出我固件内集成的文件系统驱动。

Youkii-Chen, 2021-11-18:
那真奇了怪了，我在你的x64.config里没找到有关的选项，但是用你的方法列举已安装的kmod倒是有。

我用的是xfs的硬盘

Youkii-Chen, 2021-11-18:
这个xfs是我后来装过的，不知道是你内置的还是我自己装的

Youkii-Chen, 2021-11-18:
[image]

Youkii-Chen, 2021-11-18:
最底下这个xfs是我自己用opkg 装过一遍, 不知道是你内置的还是我自己装的. 但是我用opkg 装过之后依然是插着硬盘无法开机.

SuLingGG, 2021-11-18:
> 最底下这个xfs是我自己用opkg 装过一遍, 不知道是你内置的还是我自己装的. 但是我用opkg 装过之后依然是插着硬盘无法开机.

我的固件里确实没加 xfs 驱动。
不过讲真...
xfs 的文件系统也不算常用的文件系统啊 (挠头

Youkii-Chen, 2021-11-18:
阿这，现在xfs不是挺常见的吗😂😂

SuLingGG, 2021-11-18:
> 阿这，现在xfs不是挺常见的吗😂😂

一个 OpenWrt 我支持 xfs 干啥...

Youkii-Chen, 2021-11-18:
雀拾，那我转成其他格式吧，不过ext4性能雀拾不好，用xfs比较快一点

closed 2021-11-18 by Youkii-Chen