SuLingGG/OpenWrt-Rpi #264: 树莓派 1B+无法启动，一直循环自动重启

YsMilan, 2021-11-22:
刷入固件，启动时显示:

15659008 mmcblk0
3.7357851 driver: mmcblk
3.763360] Kernel panic- not syncing: VFS: Unable to mount root fs on unknown-block(179,2)
3.782199] CPU: 0 PID: 1 Comm: swapper Not tainted 5.4.155 10
3.7982051 Hardware name: BCM2835
3.8118081 L<c010e600>] (unwind_backtrace) from [(c010b6b4>1 (show stack+0x10/0x14)
3.8297361 (<c01066b4>] (show_stack) from (‹c011de24>] (panic+Oxdc/0x314)
3.8467701 (<c011de24>1 (panic) from [<c0901328>] (mount_block_root+0x248/0x320)
3.8644331 (<c0901328>1 (mount _block_root) from [‹c090172c›l (prepare_namespace+@x158/0x1b0)
3.883155] (<090172c>1 (prepare_namespace) from [<c06ac8d4>] (kernel_init+0x8/0x110)
3.9013071 (<c06ac8d4>1 (kernel_init) from ((c01010e8>] (ret _from_fork+0x14/0x2c)
3.9191771 Exception stack (Oxd744ffbo to Oxd744fff8)
3.9344351 ffa0:
00000000 00000000 00000000 00000000
3.9530381 ffco: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
3.9714921 ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
3.9982291 Rebooting in 1 seconds.

之后自动重启，然后一直循环

SuLingGG, 2021-11-22:
树莓派 1 是很老的设备了，我这边没有条件测试。
有条件的话换张 SD 卡，或者按这里的方法 <https://openwrt.cc/restore/> 操作，重新刷一遍固件试试。

YsMilan, 2021-11-24:
> 树莓派 1 是很老的设备了，我这边没有条件测试。 有条件的话换张 SD 卡，或者按这里的方法 https://openwrt.cc/restore/ 操作，重新刷一遍固件试试。

不行...不管它了，本来也是闲置的想刷上玩玩，谢谢了